indilib/indi #9: Travis-CI config

jochym, 2016-05-29:
1. I have built a basic continuous integration setup with Travis-CI. At the moment it runs and tests most builds on linux. It is triggered on each pull request and the result is visible on github - so we can see if the patch breaks the build or not. I will probably add debian building tests later. But I know nothing about OSX build procedure. Travis has OSX images as well as linux. I think @polakovic has some experience with OSX. Would you help me with this?
 2. Currently it builds the core all libraries and most of the drivers. I want this to be as complete as possible. @knro would you check, in a spare moment, if I missed some drivers?
 3. Tests. At the moment there are no tests. It only builds. Anyone have any idea for what tests should we run to check the build?

polakovic, 2016-05-29:
OSX build is generated by XCode project now, "cmake" build is abandoned, there is no substantial demand for such solution by Mac users. But I'll look at Travis.

jochym, 2016-05-29:
There is xcode environment at travis so we can use a native toolchain on OSX - otherwise that would be  quite pointless.

polakovic, 2016-05-29:
OK, I have no experience yet, but will look at it.

polakovic, 2016-05-30:
So I looked at it and I think that we'll not use it for OSX port now. The project is quite complex, it contains not just open source, but also closed source codes and also some dependencies like cfitsio and libusb are part of this xcode project and built from source. It will be very hard to maintain it for Travis and added value is quite low, because probably nobody but me will be able to fix any issues related to OSX packaging.

jochym, 2016-05-30:
OK. Clear. So the issue for OSX is closed. Thanks @polakovic .

A-j-K, 2016-06-07:
http://indilib.org/forum/general/1315-indi-testing-framework.html?limitstart=0

I'm not sure what the best place to discuss this is, here or the forum. My personal preference is here as this is where the code is.

A-j-K, 2016-06-12:
> Tests. At the moment there are no tests. It only builds. Anyone have any idea for what tests should we run to check the build?

I've been talking with @knro about testing in general and have started some work here:-

https://github.com/A-j-K/indi/tree/feature/unittest

https://travis-ci.org/A-j-K/indi

More info to come as I progress.

A-j-K, 2016-06-21:
https://github.com/indilib/indi/pull/35

This PR does some work with the Travis-CI and brings UTs to the end of the build. I am guessing this will advance as we go in the UT project we are doing.

closed 2017-03-04 by knro