indilib/indi #88: [EKOS] Light_1000.fits gets overwritten again and again without warning

d33psky, 2016-10-16:
Light_1000.fits gets overwritten again and again. Even worse: without warning.

kstars-bleeding 5:16.10+201610151447~ubuntu16.04.1

lupinix, 2016-10-16:
Unfortunately a known issue https://bugs.kde.org/show_bug.cgi?id=348880

d33psky, 2016-10-16:
Oh ! Well, that fix worked for about 1 year ;-)
I saw your proposed patch and Jasem's remark that it took more than just that.
Maybe it's an idea to fix this issue up to the point where increasing it even further makes no sense filesystem wise.
Jasem: which was the actual commit that upped the limit from 100 to 999 ?

knro, 2016-10-16:
This is not an INDI issue so closing it here, please follow up with bugs.kde.org and I'll try to find a solution.

closed 2016-10-16 by knro