indilib/indi #107: What does telescope parking means and how to toggle it?

4l1fe, 2016-11-11:
Hi, i have a python custom client and a mount i try to set direction and slew rate in.
But log tell me that mount is parked. I try to find a way to unpark it.
Look at an approximate code:
```python
        switch_vec = self.indi_client.telescope.getSwitch('TELESCOPE_PARK')
        if (switch_vec[0].s, switch_vec[1].s) != (c.OFF, c.ON):
            switch_vec[0].s, switch_vec[1].s = c.OFF, c.ON
            self.indi_client.sendNewSwitch(switch_vec)
            logger.debug('Sent park off')

        switch_vec = self.indi_client.telescope.getSwitch('TELESCOPE_SLEW_RATE')
        for switch in switch_vec:
            switch.s = c.OFF
        if r:
            switch_vec[r-1].s = c.ON
        self.indi_client.sendNewSwitch(switch_vec)

        switch_vec = self.indi_client.telescope.getSwitch(property_)
        if (switch_vec[0].s, switch_vec[1].s) != self.DIRECTIONS[d]['states']:
            switch_vec[0].s, switch_vec[1].s = self.DIRECTIONS[d]['states']
            self.indi_client.sendNewSwitch(switch_vec)
            logger.debug('Sent new direction')
```
These switches set correctly but mount hasn't moved.
So in addition to the topic questions is there a rules description how to set protperties for different devices in a right way?
I haven't fount it in the indi protocol and indilib.org.

Help a bit please.

knro, 2016-11-13:
Run the mount under indiserver -vvv and see what XML messages gets communicated. You should see the problem there. The code looks OK if I'm understanding it correctly. But find out what gets exactly transmitted to the driver from the -vvv verbose XML first.

4l1fe, 2016-11-14:
> Another thing you could try is to wait until the mount acknowledge 'unparking' OK before proceeding with the rest of the code. You can do that by checking the status the of the TELESCOPE_PARK switch, it should become IPS_OK when it's done.
Yea, i did so. My mistake was setting little slew rate to note an effect. With the max rate it works as expected after unparking. I should read logs better.

But i don't still understand what parking is for, its semantics, its appointment.
So thank you anyway.

closed 2016-11-14 by 4l1fe