indilib/indi #114: indi-asi always shows a second camera

d33psky, 2016-11-18:
A second ZWO CCD camera pops up with indi-asi 0.5+r2732~201611180003~ubuntu16.04.1

indi_getprop shows it as :
```
ZWO CCD �.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
ZWO CCD �.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD �.ACTIVE_DEVICES.ACTIVE_SKYQUALITY=SQM
ZWO CCD �.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD �.CONFIG_PROCESS.CONFIG_DEFAULT=Off
ZWO CCD �.CONFIG_PROCESS.CONFIG_LOAD=Off
ZWO CCD �.CONFIG_PROCESS.CONFIG_SAVE=Off
ZWO CCD �.CONNECTION.CONNECT=Off
ZWO CCD �.CONNECTION.DISCONNECT=On
ZWO CCD �.DEBUG.DISABLE=On
ZWO CCD �.DEBUG.ENABLE=Off
ZWO CCD �.DRIVER_INFO.DRIVER_EXEC=indi_asi_ccd
ZWO CCD �.DRIVER_INFO.DRIVER_INTERFACE=2
ZWO CCD �.DRIVER_INFO.DRIVER_NAME=ZWO CCD
ZWO CCD �.DRIVER_INFO.DRIVER_VERSION=0.5
ZWO CCD �.SIMULATION.DISABLE=On
ZWO CCD �.SIMULATION.ENABLE=Off
```
with the regular camera also pasted for completeness : 
```
ZWO CCD ASI1600MM-Cool.ACTIVE_DEVICES.ACTIVE_FILTER=EFW
ZWO CCD ASI1600MM-Cool.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI1600MM-Cool.ACTIVE_DEVICES.ACTIVE_SKYQUALITY=SQM
ZWO CCD ASI1600MM-Cool.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Celestron GPS
ZWO CCD ASI1600MM-Cool.CONFIG_PROCESS.CONFIG_DEFAULT=Off
ZWO CCD ASI1600MM-Cool.CONFIG_PROCESS.CONFIG_LOAD=Off
ZWO CCD ASI1600MM-Cool.CONFIG_PROCESS.CONFIG_SAVE=Off
ZWO CCD ASI1600MM-Cool.CONNECTION.CONNECT=Off
ZWO CCD ASI1600MM-Cool.CONNECTION.DISCONNECT=On
ZWO CCD ASI1600MM-Cool.DEBUG.DISABLE=Off
ZWO CCD ASI1600MM-Cool.DEBUG.ENABLE=On
ZWO CCD ASI1600MM-Cool.DEBUG_LEVEL.DBG_DEBUG=On
ZWO CCD ASI1600MM-Cool.DEBUG_LEVEL.DBG_ERROR=On
ZWO CCD ASI1600MM-Cool.DEBUG_LEVEL.DBG_SESSION=On
ZWO CCD ASI1600MM-Cool.DEBUG_LEVEL.DBG_WARNING=On
ZWO CCD ASI1600MM-Cool.DRIVER_INFO.DRIVER_EXEC=indi_asi_ccd
ZWO CCD ASI1600MM-Cool.DRIVER_INFO.DRIVER_INTERFACE=2
ZWO CCD ASI1600MM-Cool.DRIVER_INFO.DRIVER_NAME=ZWO CCD
ZWO CCD ASI1600MM-Cool.DRIVER_INFO.DRIVER_VERSION=0.5
ZWO CCD ASI1600MM-Cool.LOGGING_LEVEL.LOG_DEBUG=On
ZWO CCD ASI1600MM-Cool.LOGGING_LEVEL.LOG_ERROR=On
ZWO CCD ASI1600MM-Cool.LOGGING_LEVEL.LOG_SESSION=On
ZWO CCD ASI1600MM-Cool.LOGGING_LEVEL.LOG_WARNING=On
ZWO CCD ASI1600MM-Cool.LOG_OUTPUT.CLIENT_DEBUG=On
ZWO CCD ASI1600MM-Cool.LOG_OUTPUT.FILE_DEBUG=On
ZWO CCD ASI1600MM-Cool.SIMULATION.DISABLE=On
ZWO CCD ASI1600MM-Cool.SIMULATION.ENABLE=Off
```

This is on Ubuntu 16.04.1 , 4.4.0-47-generic kernel , latest INDI from bleeding PPA with indi-asi 0.5+r2732~201611180003~ubuntu16.04.1

rumengb, 2016-11-19:
ZWO changed some things under the hood to enable camera hot plug support for our indigo project and fixes some other issues. There is a chance that the proper way of calling of some functions changed. I will look in to this.

rumengb, 2016-11-19:
I tried it with 2 cameras:
 
2 cameras connected:
rumen@home-desktop:~$ indi_getprop 
ZWO CCD ASI224MC.CONNECTION.CONNECT=Off
ZWO CCD ASI224MC.CONNECTION.DISCONNECT=On
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_NAME=ZWO CCD
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_EXEC=indi_asi_ccd
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_VERSION=0.5
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_INTERFACE=2
ZWO CCD ASI224MC.DEBUG.ENABLE=Off
ZWO CCD ASI224MC.DEBUG.DISABLE=On
ZWO CCD ASI224MC.SIMULATION.ENABLE=Off
ZWO CCD ASI224MC.SIMULATION.DISABLE=On
ZWO CCD ASI224MC.CONFIG_PROCESS.CONFIG_LOAD=Off
ZWO CCD ASI224MC.CONFIG_PROCESS.CONFIG_SAVE=Off
ZWO CCD ASI224MC.CONFIG_PROCESS.CONFIG_DEFAULT=Off
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
ZWO CCD ASI071MC-Cool.CONNECTION.CONNECT=Off
ZWO CCD ASI071MC-Cool.CONNECTION.DISCONNECT=On
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_NAME=ZWO CCD
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_EXEC=indi_asi_ccd
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_VERSION=0.5
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_INTERFACE=2
ZWO CCD ASI071MC-Cool.DEBUG.ENABLE=Off
ZWO CCD ASI071MC-Cool.DEBUG.DISABLE=On
ZWO CCD ASI071MC-Cool.SIMULATION.ENABLE=Off
ZWO CCD ASI071MC-Cool.SIMULATION.DISABLE=On
ZWO CCD ASI071MC-Cool.CONFIG_PROCESS.CONFIG_LOAD=Off
ZWO CCD ASI071MC-Cool.CONFIG_PROCESS.CONFIG_SAVE=Off
ZWO CCD ASI071MC-Cool.CONFIG_PROCESS.CONFIG_DEFAULT=Off
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator

removed one  camera: 
rumen@home-desktop:~$ indi_getprop 
ZWO CCD ASI071MC-Cool.CONNECTION.CONNECT=Off
ZWO CCD ASI071MC-Cool.CONNECTION.DISCONNECT=On
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_NAME=ZWO CCD
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_EXEC=indi_asi_ccd
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_VERSION=0.5
ZWO CCD ASI071MC-Cool.DRIVER_INFO.DRIVER_INTERFACE=2
ZWO CCD ASI071MC-Cool.DEBUG.ENABLE=Off
ZWO CCD ASI071MC-Cool.DEBUG.DISABLE=On
ZWO CCD ASI071MC-Cool.SIMULATION.ENABLE=Off
ZWO CCD ASI071MC-Cool.SIMULATION.DISABLE=On
ZWO CCD ASI071MC-Cool.CONFIG_PROCESS.CONFIG_LOAD=Off
ZWO CCD ASI071MC-Cool.CONFIG_PROCESS.CONFIG_SAVE=Off
ZWO CCD ASI071MC-Cool.CONFIG_PROCESS.CONFIG_DEFAULT=Off
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI071MC-Cool.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator

connected the other camera and connected the first:
rumen@home-desktop:~$ indi_getprop 
ZWO CCD ASI224MC.CONNECTION.CONNECT=Off
ZWO CCD ASI224MC.CONNECTION.DISCONNECT=On
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_NAME=ZWO CCD
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_EXEC=indi_asi_ccd
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_VERSION=0.5
ZWO CCD ASI224MC.DRIVER_INFO.DRIVER_INTERFACE=2
ZWO CCD ASI224MC.DEBUG.ENABLE=Off
ZWO CCD ASI224MC.DEBUG.DISABLE=On
ZWO CCD ASI224MC.SIMULATION.ENABLE=Off
ZWO CCD ASI224MC.SIMULATION.DISABLE=On
ZWO CCD ASI224MC.CONFIG_PROCESS.CONFIG_LOAD=Off
ZWO CCD ASI224MC.CONFIG_PROCESS.CONFIG_SAVE=Off
ZWO CCD ASI224MC.CONFIG_PROCESS.CONFIG_DEFAULT=Off
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
ZWO CCD ASI224MC.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator

Can you please be more descriptive and provide steps how to reproduce the issue?
By the way did you upgrade the indiserver and indilib together with indi-asi?

d33psky, 2016-11-19:
I have only 1 physical ZWO CCD camera.
The problem is that 2 ZWO CCD show up in INDI.

I start indiserver as follows : 
```
gdb --ex=run --eval-command="set follow-fork-mode child" --args indiserver -v -p 7624 -m 100 indi_asi_wheel indi_asi_ccd
```

And then ran : 
```
indi_getprop | sort -u
```
which output I pasted above when I made this bug ticket.

indiserver and indilib incl indi-asi were all upgraded together from kstars-bleeding PPA.

rumengb, 2016-11-19:
I ran the server the same way through gdb. Without understanding why it happens with gdb and why would one run indi trough gdb and see a second camera. However I did not see a second camera. Sorry, but you will have to write a bit more detailed description of what you do if you want  this to be solved :) Do not be shy! :)
Any additional USB devices connected? There may be some other device interpreted as ASI camera... 
Additionally can you paste here the output of "lsusb", but without running it through gdb please :)

Rumen

d33psky, 2016-11-19:
Here's some clue : it is the presence of the ZWO EFW.
With this filter wheel plugged in indilib always shows 2 cameras.

I've upgraded to indi-asi version
```
0.5+r2733~201611190003~ubuntu16.04.1
```
which gives the same wrong result. Just so that you know I also tested with a later version.

I've unplugged all other astro USB devices. Present are : ccd, efw and a mouse : 
```
lsusb
Bus 002 Device 010: ID 03c3:1f01
Bus 002 Device 009: ID 03c3:1603
Bus 002 Device 008: ID 04b4:6572 Cypress Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 002: ID 046d:c050 Logitech, Inc. RX 250 Optical Mouse
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 0a5c:5800 Broadcom Corp. BCM5880 Secure Applications Processor
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

gdb is used to debug indi filter wheel driver which had issues before. using gdb is described here : 
http://www.indilib.org/forum/general/571-read-before-you-post-any-support-request.html

rumengb, 2016-11-19:
Figured it out. This is a bug in the ASI SDK. EFW is threated as camera. I have sent report to Yang. he will fix the SDK. It usually takes 1-2 days and I will update the driver once he is ready.

Thank you!

Rumen

rumengb, 2016-11-22:
Issue should be fixed with my last commit. created pull request.

closed 2016-11-22 by knro