indilib/indi #115: indi-asi no longer allows fan control

d33psky, 2016-11-18:
Ubuntu 16.04.1 , 4.4.0-47-generic kernel , latest INDI from bleeding PPA with indi-asi version ``` 0.5+r2732~201611180003~ubuntu16.04.1 ```
no longer allows FAN control.

The tab 'Controls' where USB bandwidth can be set no longer has an entry to turn the fan on or off.

Can this be restored please ? Thanks.

rumengb, 2016-11-19:
This was deliberately disabled by ZWO, I asked them several days ago. The explanations was that turning it off causes problems with some cameras.

d33psky, 2016-11-19:
Damn that is really bad news :(
What is the last asi-indi version that did support fan control ? I need to rollback to that version :-/

rumengb, 2016-11-19:
Excuse my ignorance, but why is fan control that important? my 1600 was not even wired to be able to turn off the fan. Any version before 15 Nov will have fan control and probably will not show the EFW wheel as camera.

d33psky, 2016-11-19:
I keep my equipment powered on 24h a day, and this little fan will dust up and die on me in no time if I allow it to run 24h a day.
ps. I do disable the piezo cooling when not imaging. If the fan can be tied to that I'm OK too.

rumengb, 2016-11-22:
FAN is turned on and off when the cooler is turned on and off now. created pull request.
This should be ok for you :)

closed 2016-11-22 by knro