indilib/indi #132: indi_sx_ccd segfaults on Ubuntu 16.04 LTS

teuben, 2016-12-23:
**Starting from the current git repo I'm starting the indiserver to talk to my Starlight Express: **

indiserver -vvv indi_sx_ccd 

**shows in /var/log/kern.log that indi_sx_ccd died:**

Dec 22 19:52:05 nemo2 kernel: [21577.370424] usb 3-1: reset high-speed USB device number 15 using xhci_hcd
Dec 22 19:52:05 nemo2 kernel: [21577.500185] indi_sx_ccd[13693]: segfault at 40 ip 00007fa6aba952a6 sp 00007ffd4928ffe0 error 4
Dec 22 19:52:05 nemo2 kernel: [21577.500191] usb 3-1: USB disconnect, device number 15
Dec 22 19:52:05 nemo2 kernel: [21577.500199]  in libusb-1.0.so.0.1.0[7fa6aba8d000+17000]
Dec 22 19:52:05 nemo2 kernel: [21577.666614] usb 3-1: new high-speed USB device number 16 using xhci_hcd

**Then executing the sx_ccd_test program showed a bit more.  Running it through the debugger, showed it dies in the libusb-1.0 library:**

(gdb) n
338	  int rc = libusb_bulk_transfer(sxHandle, BULK_OUT, setup_data, 8, &transferred, BULK_COMMAND_TIMEOUT);
(gdb) s

Thread 1 "sx_ccd_test" received signal SIGSEGV, Segmentation fault.
0x00007ffff7bc72aa in libusb_submit_transfer () from /lib/x86_64-linux-gnu/libusb-1.0.so.0

**As a next step I might compile the USB library to see if this gives me more insight, but obviously would be greatful for any other hints on this problem on which I've been banging for a few days now.**

teuben, 2016-12-23:
After some more testing, I should add that the segfault is not persistent. Re-executing  **sx_ccd_test** will actually work, and I also get the impression a time-out within things work or fail is of relevance.

knro, 2017-03-04:
Any update on this issue?

teuben, 2017-03-04:
none, but I am moving my code to a raspberry pie as well, and hope it will fare better there. I saw there were some updates overall, so it's time to test again.

knro, 2017-03-20:
Ok can this issue be closed or still pending?

teuben, 2017-03-20:
Im working on it this next few days with the camera in haiti. Will update
shortly, as im also going to.use the new 1.4.1 version.

knro, 2017-04-15:
If there are no more updates, I'll be closing this issue soon.

closed 2017-05-17 by knro