indilib/indi #137: Compile error from the last github src

TallFurryMan, 2017-01-03:
Build problem originally posted at http://indilib.org/forum/general/1779-compile-error-from-the-last-github-src.html

The errors are reported on distribution raspbian, and are similar to:
> /home/pi/Projects/indi/libindi/libs/webcam/v4l2_base.h: In member function ‘bool V4L2_Base::is_compressed() const’:
> /home/pi/Projects/indi/libindi/libs/webcam/v4l2_base.h:174:51: error: ‘const struct v4l2_pix_format’ has no member named ‘flags’
>   bool is_compressed() const { return fmt.fmt.pix.flags & V4L2_FMT_FLAG_COMPRESSED; };

Apparently latest raspbian kernel headers [exact version missing] doesn't provide headers declaring pixel format flags.

No issue on Ubuntu Mate kernel headers [exact version missing].
No issue on Gentoo, kernel headers 4.4.

TallFurryMan, 2017-01-04:
Field "flags" was introduced in kernel 3.17.
Jasem, I saw your changes in f955aa, when should I use DEBUGFDEVICE instead of IDLog? Do we rely on the client's log settings only?

knro, 2017-01-04:
Yes DEBUG & friends should be used when logging whenever possible as it is
part of the INDI Logging framework.

commit 4d2fe1a2 (2017-01-05): Fixes #137 - compile issue in V4L2 client with 3.x kernels.

Make sure source compatibility is preserved with 3.x kernels:
- Field 'flags' is available in the V4L2 API starting kernel 3.17,
- Timestamps were introduced in the V4L2 API starting kernel 3.15,
- DMABUF was introduced in the V4L2 API starting kernel 3.8.

Tested with kernel headers 3.6, 3.8, 3.9, 3.16, 3.17, 3.19, 4.4.
commit 94126c9d (2017-01-05): Merge pull request #138 from TallFurryMan/bugfix_v4l2_3x_kernel_headers

Fixes #137 - compile issue in V4L2 client with 3.x kernels.
closed 2017-01-05 by knro