indilib/indi #146: Loading sequence in capture module crashes

jpaana, 2017-01-30:
With both current PPA (`kstars-bleeding 5:16.12+201701262218~ubuntu16.04.1, indi-full 1.3~201701301122~ubuntu16.04.1`) and self-compiled git version of kstars I get a crash when trying to load a sequence file. Running on current Ubuntu 16.04 LTS on amd64. Reproduction in kstars:
- start local indi with simulators and connect (I used eqmod mount driver simulator, but it doesn't seem to matter)
- go to capture module and press load sequence button
- "Connection to INDI host at localhost on port 7624 lost. Server disconnected." dialog pops up.
- file dialog is operable and if I select a file, I get this backtrace in gdb:
`#0  0x00007ffff4b7eecf in QTableWidget::rowCount() const ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x000000000075bfc9 in Ekos::Capture::loadSequenceQueue (this=0x49f6190, 
    fileURL=...)
    at /home/jarno/Projects/kstars/kstars/ekos/capture/capture.cpp:2282
#2  0x000000000075be37 in Ekos::Capture::loadSequenceQueue (this=0x49f6190)
    at /home/jarno/Projects/kstars/kstars/ekos/capture/capture.cpp:2262
#3  0x000000000061cbd9 in Ekos::Capture::qt_static_metacall (_o=0x49f6190, 
    _c=QMetaObject::InvokeMetaMethod, _id=49, _a=0x7fffffffce00)
    at /home/jarno/Projects/build/kstars/kstars/moc_capture.cpp:494
#4  0x00007ffff3b0cd2a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff4c54192 in QAbstractButton::clicked(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff49b6f04 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff49b8519 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff49b8694 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff48fcf88 in QWidget::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff48ba05c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff48bfc19 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff3ade38b in QCoreApplication::notifyInternal(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff48beb32 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff491757b in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff4919b3b in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff48ba05c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff48bf516 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff3ade38b in QCoreApplication::notifyInternal(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff43094e1 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#20 0x00007ffff430b1a5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#21 0x00007ffff42eef08 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#22 0x00007fffe4bb5060 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#23 0x00007fffeecdf197 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007fffeecdf3f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007fffeecdf49c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007ffff3b347cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff3adbb4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff3ae3bec in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x000000000048bd9d in main (argc=1, argv=0x7fffffffdde8)
    at /home/jarno/Projects/kstars/kstars/main.cpp:278
`

jpaana, 2017-01-30:
On my Raspberry Pi 3 loading seems to work, but it's running slightly older PPA (`kstars-bleeding 5:16.12+201701262218~ubuntu16.04.1 and indi-full 1.3~201701281703~ubuntu16.04.1`).

jpaana, 2017-01-30:
Updated Raspberry packages, indi-full is now `1.3~201701301122~ubuntu16.04.1` and loading still works, kstars-bleeding is still the same `5:16.12+201701262218~ubuntu16.04.1`

knro, 2017-01-30:
Can't reproduce this. Can you attach the sequence file you were trying to load?

jpaana, 2017-01-31:
Connection to indi server is lost already before selecting any sequence file, just pressing the load button. I also tried with a locally run "remote" indi server and it seems the server is fine and continues sending mount updates and so on even after kstars says connection lost. In any case I'll attach the particular sequence file, though it seems to do the same with any of them I've tried.
[seq.zip](https://github.com/indilib/indi/files/741352/seq.zip)

jpaana, 2017-01-31:
Similar error also happens if I try to save a new sequence on this machine. Indi server disconnect when I press the button and crash after selecting a file name. It does create a file with contents
`<?xml version="1.0" encoding="UTF-8"?>
<SequenceQueue version='1.5'>`

Backtrace is quite similar to the load case:
`#0  0x00007ffff3a31efc in QUrl::~QUrl() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x000000000075d574 in Ekos::Capture::saveSequenceQueue (this=0x4333520)
    at /home/jarno/Projects/kstars/kstars/ekos/capture/capture.cpp:2560
#2  0x000000000075dac8 in Ekos::Capture::saveSequenceQueueAs (this=0x4333520)
    at /home/jarno/Projects/kstars/kstars/ekos/capture/capture.cpp:2605
#3  0x000000000061cb4f in Ekos::Capture::qt_static_metacall (_o=0x4333520, 
    _c=QMetaObject::InvokeMetaMethod, _id=51, _a=0x7fffffffce10)
    at /home/jarno/Projects/build/kstars/kstars/moc_capture.cpp:496
#4  0x00007ffff3b0cd2a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff4c54192 in QAbstractButton::clicked(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff49b6f04 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff49b8519 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff49b8694 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff48fcf88 in QWidget::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff48ba05c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff48bfc19 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff3ade38b in QCoreApplication::notifyInternal(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff48beb32 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
`
and so on.

jpaana, 2017-01-31:
PPA gives similar backtrace when saving, line numbers are slightly different compared to git head:
`
#0  0x00007ffff3a31efc in QUrl::~QUrl() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00000000006ae12d in Ekos::Capture::saveSequenceQueue (this=0x4772880)
    at /build/kstars-bleeding-DRDwgu/kstars-bleeding-16.12+201701262218~ubuntu16.04.1/kstars/ekos/capture/capture.cpp:2550
#2  0x00000000006ae589 in Ekos::Capture::saveSequenceQueueAs (
    this=<optimized out>)
    at /build/kstars-bleeding-DRDwgu/kstars-bleeding-16.12+201701262218~ubuntu16.04.1/kstars/ekos/capture/capture.cpp:2595
#3  0x00000000005b2b1f in Ekos::Capture::qt_static_metacall (
    _o=<optimized out>, _c=<optimized out>, _id=<optimized out>, 
    _a=<optimized out>)
    at /build/kstars-bleeding-DRDwgu/kstars-bleeding-16.12+201701262218~ubuntu16.04.1/obj-x86_64-linux-gnu/kstars/moc_capture.cpp:496
#4  0x00007ffff3b0cd2a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff4c54192 in QAbstractButton::clicked(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
`

knro, 2017-02-06:
Well I can't reproduce this on any machine, this includes VM machines with different distros as well.

jpaana, 2017-02-06:
Yes, I too can only reproduce it on that one machine, another amd64 with Ubuntu 16.10 and Raspberry Pi 3 work fine. Any idea where I should break to see what causes the indi server disconnection? My hypothesis is that the disconnect is the root cause which makes Ekos close the capture tab as it doesn't have cameras anymore and that causes then the actual crash when completing the dialog which would be caused by accessing deleted objects. The disconnect is only on Kstars side as the server remains up and sending snoop information if run as a separate process and not from Kstars directly.

closed 2017-02-26 by knro