indilib/indi #149: unable to capture using a Sony ILCE-6000 (aka Alpha-6000)

justanenduser, 2017-02-02:
I am trying to control a Sony ILCE-6000 (aka Alpha-6000) mirrorless with libindi. 

With the camera set to MANUAL & BULB I do not get the sound of a shutter release and I get this from DEBUG...

INFO	29.534555 sec	: Session log file /home/gary/.indi/logs/2017-02-02/indi_gphoto_ccd/indi_gphoto_ccd_01:59:51.log
DEBUG	30.137160 sec	: Toggle Logging Level -- Driver Debug

--the lines between the ones above and ones below repeat multiple times (depending on how I wait before closing the camera connection)--

DEBUG	52.953428 sec	: Starting exposure (exptime: 2 secs, mirror lock: 0)
DEBUG	52.953436 sec	:   Mutex locked
DEBUG	52.953444 sec	: Setting radio/menu widget iso: 11 (800)
DEBUG	52.959676 sec	: Setting new configuration OK.
DEBUG	52.959712 sec	: Opening remote serial shutter port: /dev/ttyUSB1 ...
DEBUG	52.959744 sec	: Failed to open serial port: /dev/ttyUSB1
ERROR	52.959751 sec	: Error starting exposure

With the camera set to MANUAL & 2 SEC EXPOSURE I get the sound of a shutter release and I get this from DEBUG...

INFO	52.117115 sec	: Session log file /home/gary/.indi/logs/2017-02-02/indi_gphoto_ccd/indi_gphoto_ccd_01:36:22.log
DEBUG	52.818534 sec	: Toggle Logging Level -- Driver Debug
DEBUG	53.820642 sec	: Toggle Debug Level -- Driver Debug

--the lines between the ones above and ones below repeat multiple times (depending on how I wait before closing the camera connection)--

DEBUG	84.125158 sec	: Starting exposure (exptime: 2 secs, mirror lock: 0)
DEBUG	84.125243 sec	:   Mutex locked
DEBUG	84.125273 sec	: Setting radio/menu widget iso: 11 (800)
DEBUG	84.139659 sec	: Setting new configuration OK.
DEBUG	84.139754 sec	: Opening remote serial shutter port: /dev/ttyUSB1 ...
DEBUG	84.139829 sec	: Failed to open serial port: /dev/ttyUSB1
ERROR	84.139855 sec	: Error starting exposure

I am using the INDI Client panel to CONNECT, LOAD saved settings, enable DEBUG and initiate the exposure. 

I am using libindi from GitHub obtained 2/1/2017 and libgphoto2 from GitHub obtained 1/31/2017. I am running 64 bit Fedora 25 with the updates-testing repos enabled. 

As an aside gphoto2, also from GitHub, can successfully interface with my camera through the same libgphoto2. 

All three programs were built with gcc 6.3.1

Please let me know what I can do (hopefully) to help ... thanks.

-gary-

knro, 2017-02-02:
Hi Gary,

Two things are required.

1. Attach complete INDI GPhoto log. Start DEBUG _before_ you click CONNECT in the INDI Control Panel so that all data including startup is logged.
2. Write the GPhoto2 command you use to capture from your camera from the command line.

TallFurryMan, 2017-02-02:
Isn't it just that your device is not on /dev/ttyUSB1 but elsewhere? You checked that probably. In the end if the device is not recognized by your distro, you probably miss a driver (MTP?).

justanenduser, 2017-02-02:
Five files attached: lsusb using -v, lsusb using -t, a dmesg fragment, a VERY large INDI debug and the gphoto2 command-line parameters. The debug was produced from KStars. I hope some of this is helpful. If not please let me know. 
[lsusb-output-ver1.txt](https://github.com/indilib/indi/files/748480/lsusb-output-ver1.txt)
[lsusb-output-ver2.txt](https://github.com/indilib/indi/files/748482/lsusb-output-ver2.txt)
[dmesg-output.txt](https://github.com/indilib/indi/files/748486/dmesg-output.txt)
[indi_gphoto_ccd_14:40:36.txt](https://github.com/indilib/indi/files/748491/indi_gphoto_ccd_14.40.36.txt)
[gphoto2-commandline-parameters.txt](https://github.com/indilib/indi/files/748494/gphoto2-commandline-parameters.txt)

justanenduser, 2017-02-02:
Sorry, I closed this by mistake.

justanenduser, 2017-02-05:
Some of my problem was apparently caused by my (mis)use of testing repos. I apologize. I have gone back to the distro repos.

TallFurryMan, 2017-02-06:
Too bad we could not help on this one. Could you share additional comments on what caused the issue in the end?

knro, 2017-02-06:
Is this issue resolved or not? Also, the INDI Log file is odd, it keeps repeating the same information? It appears that if you are using the _latest_ libgphoto2 then Sony is supported.

justanenduser, 2017-02-06:
My humble apologies. Please feel free to close this. My difficulty is
apparently with the way libgphoto2 interacts with my camera.

-gary-

closed 2017-02-10 by justanenduser