indilib/indi #158: Unable to connect USB devices to INDI with RPi3 and iAstroHub v3

HxPI, 2017-02-19:
Flip Flat and Focuslynx focus controller are plugged directly into USB hub of RPi3. INDI drivers for a Flip Flat and FocusLynx focus controller are selected and INDI server is started. Using local and remote INDI control, going to the main settings for each device and selecting "Connect" causes the status to turn Red and the connection fails. I haven't tried LSUSB yet but it seems as though these devices are either not being recognized properly on the USB bus or are not being properly handled between the RPi3 and INDI. I have no problem using these devices with INDI on MBP laptop.

knro, 2017-02-19:
INDI Library version?
output of **lsusb**?
outfor of **cat /lib/udev/rules.d/99-flipflat.rules**

HxPI, 2017-02-19:
Please see attached information
[Distro Name.rtfd.zip](https://github.com/indilib/indi/files/785966/Distro.Name.rtfd.zip)

knro, 2017-02-20:
Looks you have two USB Serial devices. When you connect to flipflat, what do you use for the Port? You should try both /dev/ttyUSB0 and if that doesn't work then /dev/ttyUSB1

Because they get randomly assigned as ttyUSB0 or ttyUSB1 depending on which was detected first by the system.

knro, 2017-02-20:
So if flipflat is /dev/ttyUSB0 then Focus Boss port is /dev/ttyUSB1 and vice-versa

knro, 2017-03-04:
Any update on this issue?

HxPI, 2017-03-19:
I am still having issues with using my Flip Flat, Focuslynx, and SX USB filter wheel through external hubs connected to the RPi3. I tried to verify the ports but the issue remains.

The Flip Flat, Focuslynx, and SX USB filter wheel are mounted on top of the scope, obviously. I cannot mount the computing device on top of the mount, mainly due to space and other connectivity considerations, so connecting the devices directly is not an option. Using ssh, "lsusb" shows several devices with no information listed next to them. Going into details with "lsusb -v" shows the USB hubs are recognized but it's unclear whether the devices in question are being recognized. For the Linux uninitiated, this is a little daunting to have to work through!
 
Just for FYI, here is the "lsusb" output:
 
Bus 001 Device 040: ID 03c3:1603 
Bus 001 Device 039: ID 2109:2812 
Bus 001 Device 050: ID 2109:2812 
Bus 001 Device 042: ID 03c3:174b 
Bus 001 Device 041: ID 2109:2812 
Bus 001 Device 005: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 108: ID 0781:5150 SanDisk Corp. SDCZ2 Cruzer Mini Flash Drive (thin)
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
 
The RPi3 is so nice as a low power and portable astronomy computer solution and is definitely worth giving it a try. Unfortunately I am going to have to find another solution. I'm sure this has as much to do with my lack of Linux knowledge as the constrained configuration of my equipment. I think this situation would be much easier to diagnose and resolve if there were a more user friendly interface to work with the software/hardware. I will continue to monitor developments for this device and think of ideas and see if there is a way to get these devices working with the RPi3 but right now this device is not doing what I need it to do. I'll get an Intel Compute Stick, where I know everything will work for my configuration and needs!
 
I don't think this is an INDI issue so I'll close this thread. Thanks again for your support.
 
Ciao,
Mel

closed 2017-03-19 by HxPI

TallFurryMan, 2017-03-19:
Hello, I'm not clear on your exact setup, specifically for the term "hub", but you should be aware about the fact that USB hubs, when connected with a device which negotiates USB3, may switch to a USB3-only protocol, making other USB2 devices inoperant. You should probably test each device one by one to find when the problem occurs, if any. I remember posting that answer in the forum for a very similar problem.