indilib/indi #160: INDI/KStars Issues (Windows)

hawkndawgs, 2017-02-20:
When trying to connect via KStars/Ekos to my INDI Server (on Windows) my SSAG guide scope will not connect. The error I get when trying to connect it is:

2017-02-19T17:47:09.5: Retrieving the COM class factory for component with CLSID {DCF01AC2-8F59-4A0A-A858-36467E7BA582} failed due to the following error: 80040154 Class not registered (Exception from HRESULT: 0x80040154 (REGDB_E_CLASSNOTREG)).

KStars does not seem to be able to plate solve at all. When I load an image into the solver, it attempts to solve it, then fails. The errors don't seem to be consistent, I either get a job ID failure or just a message that says the solve failed.

2017-02-19T17:57:12 Failed to retrieve job ID.
2017-02-19T17:57:12 {"job_calibrations":[],"jobs":[null],"processing_finished":"2017-02-19 20:56:55.777920","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[1509564]}
2017-02-19T17:57:09 {"job_calibrations":[],"jobs":[null],"processing_finished":"2017-02-19 20:56:55.777920","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[1509564]}
2017-02-19T17:57:07 {"job_calibrations":[],"jobs":[null],"processing_finished":"2017-02-19 20:56:55.777920","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[1509564]}
2017-02-19T17:57:05 {"job_calibrations":[],"jobs":[null],"processing_finished":"2017-02-19 20:56:55.777920","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[1509564]}
2017-02-19T17:57:03 {"job_calibrations":[],"jobs":[null],"processing_finished":"2017-02-19 20:56:55.777920","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[1509564]}
2017-02-19T17:57:01 {"job_calibrations":[],"jobs":[null],"processing_finished":"2017-02-19 20:56:55.777920","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[1509564]}
2017-02-19T17:56:58 {"job_calibrations":[],"jobs":[null],"processing_finished":"2017-02-19 20:56:55.777920","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[1509564]}
2017-02-19T17:56:56 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"2017-02-19 20:56:53.341378","user":2633,"user_images":[]}
2017-02-19T17:56:54 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:52 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:50 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:47 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:45 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:43 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:41 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:39 {"job_calibrations":[],"jobs":[],"processing_finished":"None","processing_started":"None","user":2633,"user_images":[]}
2017-02-19T17:56:38 Upload complete. Waiting for astrometry.net solver to complete...
2017-02-19T17:56:38 {"hash":"9cfc6f0c542ba728bad9a78bce2110b69fad8f97","status":"success","subid":1463888}
2017-02-19T17:55:33 Uploading file...

I have tested uploading the file myself, and astrometry.net is able to solve it just fine.

knro, 2017-02-20:
Thanks for the report. I reported the INDI Server for Windows issue to the author. For the astrometry.net, I doubled the number of tries it attempts before given up. Will make another Windows release in a day or two.

closed 2017-02-20 by knro

hawkndawgs, 2017-02-20:
Thanks.

Can you add support for ANSVR?
http://adgsoftware.com/ansvr/

knro, 2017-02-20:
It's already supported. See http://knro.blogspot.com/2017/02/kstars-274-for-windows-is-released.html

Tested it on my Windows 10 64bit machine and it's working

hawkndawgs, 2017-02-20:
Thanks.

I did not try removing the API Key. Works great now!

polakovic, 2017-02-20:
The problem is, that most of ASCOM applications and drivers are 32 bit even on 64 bit OS. You probably used 64 bit wINDI against 32 bit SSAG driver.