indilib/indi #173: Some Settings are not applied to the NoIR Raspberry Pi Camera

Domi96, 2017-03-13:
I am using the Raspberry Pi with Raspian installed. On top of that I have installed the indi. The camera module is fully working with indi and without indi. I have started the indieserver:

`indiserver -v -m 100 indi_v4l2_ccd indi_gphoto_ccd`

Afterwards connected with KStars from the ekos VM. And here comes the problem: I can connect over the V4L2 CCD tab. On the Main Control page I can enter different values for exposure - the pi cam suports up to 6 seconds but for me changing values on the first page seems to make no difference.

Actually there are other settings to - for example at V4L2 Control. When changing the Brightness there it seems really change something at the the pictures. Compression Quality also changes the quality. Then there is Expsure Time, Absolute this allows values up 10000. Are these the same values that raspistill uses ? Because there you can set --shutter up to  6000000us (6000ms, 6s).

I also tried to change Auto Exporsure to Manual Mode which seems to make no difference. Another option that I have really missed is manual ISO controll. 

There is also another user with the same problems: http://indilib.org/forum/general/258-raspberry-pi-noir-camera-board.html?start=12#15101 

To prove it actualy is able to take pictures here is a moon shot from last night (there you can also see using KStars Windows Version but at the VM it has the same problems so it seems to have nothing to with the Operation System) 

[image]

 I hope anyone of you has an adivice what to do - because I have no idea what to try out anymore and I have already 3D printeda cam adapter and as you can see, sucessfully tested it.

Thanks in advance,
Dominik

knro, 2017-03-13:
Do you know of a simple open source app that runs the NoIR camera? We can look at the code and see what is it different from the INDI V4L2 driver. That's all I can say that as I do not have the camera here to investigate.

Domi96, 2017-03-13:
I have sucessfully used RPi_Cam_Web_Interface (which works fine): https://github.com/silvanmelchior/RPi_Cam_Web_Interface 

If that software is to complicate I can search for an easier one and/or run some tests if needed.

knro, 2017-10-20:
Many users are reporting issues with using the RPI Cam with INDI for a variety of reasons. We'd appreciate any contribution to to the code (V4L2 INDI driver) to make it more compatible with this camera if possible.

closed 2019-09-16 by knro