indilib/indi #179: qhy lib cannot be linked in with the driver

jochym, 2017-03-17:
The travis build for qhy driver is unable to be linked with the sdk library. See the build log (at the end):
https://travis-ci.org/jochym/indi/jobs/212140226

This is probably connected with c++11 ABI in gcc used differently in the lib and the rest of indi.
Anyone knows a work-around?

kecsap, 2017-03-18:
There is no workaround for this. Either:
- We should have a prebuilt qhy library which was compiled with the compiler flag -D_GLIBCXX_USE_CXX11_ABI=0 
- Disable qhy for old pre-C++11 compilers
- Forget old pre-C++11 compilers and disable this test in Travis.

jochym, 2017-03-18:
That is what I suspected. Additionally the problem is limited to ubuntu trusty (on most archs, couriously it seems to work on armhf https://code.launchpad.net/~jochym/+recipe/indi-qhy-daily ). The problem is it is the previous LTS ubuntu and the newest travis image. It would be nice to support it. Maybe we can keep the old sdk for the older systems ( @knro ? ). The main problem is the travis image - we can move to docker based test but it is a substantial additional work. Otherwise we need to disable the test on travis.

knro, 2017-03-18:
Using D_GLIBCXX_USE_CXX11_ABI=0 would make libqhy link in all systems including trusty?

kecsap, 2017-03-18:
@jochym It works for armhf because that prebuilt version was compiled with an old gcc (and old ABI):

GCC: (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) 4.8.3

@knro: No, looking in the prebuilt binaries:
x86_32, x86_64 and armv8 use new ABI.
armv6 uses old ABI.

The only choices:
1. Disable qhy on trusty.
2. Or drop the support for trusty.
3. Or ask the qhy guys to build binaries with -D_GLIBCXX_USE_CXX11_ABI=0 and do some CMake magic for old gcc versions.

knro, 2017-03-18:
I don't provide packages for trusty on my PPA, so looks like dropping support for trusty is the easiest path.

jochym, 2017-03-18:
I think we can draw the line at debian stable (raspian) or last centos (did not check) and last ubuntu LTS. The problem is testing it. I'll check all three tomorrow and get bask to you. As for travis we can just not test it there or move to docker image for testing.

knro, 2017-03-18:
armv6 version will compile on Raspian Jessie. I tested it today.

jochym, 2017-03-20:
I have tested the compile on debian/stable and centos7. Both are failing. centos is not a great problem since we officially support only fedora. The debian/stable is more serious since this is the only choice for stable debian distro. We can of course say that on debian we only support 1.3.1 on stable and 1.4.1 on next stable (testing=stretch) but this is sub-optimal IMHO. I am also not so sure about raspian. I'll test my system tomorrow.

kecsap, 2017-03-21:
@jochym Please share the details of the failing test environments. What compilers are available/used there? Maybe we can workaround some cases in CMake.

jochym, 2017-03-22:
The linking fails the same as on standard debian/jessie with the series of messages like this:
`/home/tester/lib/libqhy.so: undefined reference to std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string()@GLIBCXX_3.4.21`

The environment is a fairly standard install of raspbian/jessie with following sources:
`deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi firmware`
`deb http://archive.raspberrypi.org/debian/ jessie main ui`

The gcc version is 4.9.2. Any other info?

@knro Which version of debian/raspian have you used in your test ?

knro, 2017-03-22:
Can you please try again now? the armv7.bin was being used in Jessie on RPI3. I removed the file so that armv6.bin is used in both cases which works OK. Please test.

jochym, 2017-03-22:
I will, shortly (today)

jochym, 2017-03-22:
OK. I have tested it with my RPi2 system and QHY5 and it works now. 
Maybe we can use older blobs (SDK) on debian/stable? What you think? 
This would be a workable solution.

jochym, 2017-03-23:
I thought about it a bit and there seems to be no sane solution for the older (gcc4.x) distros. There is no way to link the gcc4 objects/libs with a single gcc5 library. The only way I have found (on debian/jessie) is to install gcc-6 from debian/testing and compile using it. For travis (ubuntu trusty) we can install gcc5 but it did not work at first attempt. I can fight it a little bit more later.
The question is: Is it worth the effort @knro , @kecsap ?

knro, 2017-04-15:
armv6 can be linked fine on Jessie on RPI3. So I'll be closing this issue unless there is a reason to keep it open.

closed 2017-04-15 by knro

jochym, 2017-04-17:
Well - the only reason was to keep qhy working on debian stableon the pc. It will be working again when stretch gets released (should be fairly soon) - since this is a developer release anyway and people using debian jessie are unlikely to use dev version the only remaining problem is the travis-ci build.