indilib/indi #185: sxccdtest:   NX=10 and NY=10 ???

teuben, 2017-03-22:
The code in sxccdtest is very useful to study for newcomers, but I was frustrated to see the author use a 10x10 region and not code them as NX and NY. So I'll happily volunteer to fix that, as I had been hacking in this code for other reasons.
However, printing the 10x10 region appears to print the Y along the horizontal and X along the vertical. I would normally interpret X to be horizontal and Y to be vertical, even if (0,0) is at the top left...   Any comments on this before I work on a pull request, and lift this confusion.

polakovic, 2017-03-23:
You're definitely right and feel free to do it. It was never meant as an example, just testing tool for interface library and it is far from perfect code. Peter

teuben, 2017-03-23:
It was perfect for me, except that confusion. I got a bit frustrated with other code, and with this code I was able to wrap it into an ascii PGM file, using the P2 header, followed by NX and NY and peak value, then all rows, and imagemagic convert it to a fits file, and i could see that it worked!  That inspired me to clarify the NX and NY issue. Every beginner is confused about row major and column major storage, and there's also the issue of where 0,0 even is :-)

teuben, 2017-03-23:
There are two sections where the pixels are written to cout. In the 2nd section the camIndex is 1, instead of 0. This is where only NX*NY bytes are read, instead of 2*NX*NY when camIndex is 0. That didn't seem correct to me.

polakovic, 2017-03-23:
Second section works with 8bit/pixel guider head (camindex = 1).

teuben, 2017-03-23:
thanks!   and one final query: where is the exposure time hidden? Is that the 900 ms in sxSetTimer ?  I don't have the camera with me today, but will test tomorrow.

polakovic, 2017-03-23:
In this particular case it is hidden in usleep(1000) at lines 113 and 136 - computer timed mode is used.

Camera has also built-in timer for camera timed mode for short (about 3s) exposure times. In this case you have instead of sxClearPixels/sxLatchPixels/sxReadPixels sequence use just sxExposePixels/sxReadPixels.

Actually, it is not so easy as it looks from sxccdtest.cpp, in real driver you need to measure and handle difference between odd and even frames for interlaced cameras, flush 3s before readout for long exposures etc.

polakovic, 2017-03-23:
Maybe you can also look at this:

https://github.com/indigo-astronomy/indigo/blob/master/indigo_drivers/ccd_sx/indigo_ccd_sx.c

it is slightly different approach, but it may help you to understand how to handle different camera modes.

knro, 2017-04-15:
Looks like issue is considered resolved.

closed 2017-04-15 by knro