indilib/indi #191: MGen Autoguider crashes when clicking UI buttons

TallFurryMan, 2017-03-26:
Regression introduced by commit 7d53fd.
The new conversion seems to reject the cast, while it is only integer stuff (hence I should have used integers in the first place...).

knro, 2017-03-27:
It its original form, it wouldn't compile since aux pointer was not properly used. I just pushed another change now, please check and test.

TallFurryMan, 2017-03-27:
Which compiler was used for this, so that I may spot the issue earlier myself? (I had no issue building before pushing)

knro, 2017-03-28:
I am using GCC v5.4

TallFurryMan, 2017-03-30:
Fixed by 7de115b8475cc2cd1fb3407f9087bb94881855b6, though the situation is simpler than what the fix attempts to rework. I'll get back to this maybe, just because I'm a bit obsessive.

closed 2017-03-30 by TallFurryMan