indilib/indi #194: Astrometry driver crashes when CCD frame is compressed

TallFurryMan, 2017-03-29:
Astrometry driver crashes when receiving a compressed blob. This is the header dump of the blob received by the driver:

```
2017-03-29T11:57:13: Driver indi_astrometry: <setBLOBVector device="V4L2 CCD" name="CCD1" state="Ok" timeout="60" timestamp="2017-03-29T11:27:42">
2017-03-29T11:57:13: Driver indi_astrometry:     <oneBLOB name="CCD1" size="80640" enclen="50528" format=".fits.z">
```

From there, the loop fwriting the blob to the temporary FITS file is reading past enclen, until size. It happens that sometimes the memory in that interval is readable, sometimes not. Switching Compressed to Raw in the CCD panel works around the issue. The Astrometry driver cannot use a compressed frame yet anyway, solver fails because it expects a raw one.

I'm picking the action.

TallFurryMan, 2017-03-29:
Actually I need an advice here: function dispatch() in indidriver is sending the size, not the encoded length.

The problem might be deeper than I think: when the dispatched blob is compressed, it is indidriver which needs fixing, not the Astrometry driver.

There could be many side-effects if I change indidriver.c:1173 to use attribute `enclen` instead of `size`, but I think this is how it should work.

TallFurryMan, 2017-03-29:
I got it wrong on where the error was for this exact issue, as this issue relates to a snooped device, still digging.
But the fact remains that there is a bug at indidriver.c:1173.

knro, 2017-03-29:
Oh I actually recall this issue. It was on my TODO list when I was making
the astrometry driver. I encountered many issues when developing the
astrometry driver related to how BLOBs are transferred between drivers and
fix most of them, but then I didn't proceed to uncompress the blob before
sending it off to the driver.

knro, 2017-03-29:
actually, after investigating the code a bit. I think it is better to send
blobs as is to the drivers and not assume they want to uncompress them.
Therefore, the astrometry driver should uncompress. I'll work on that.

TallFurryMan, 2017-03-30:
Fixed by b820215341c3ef95a9e6085abf8c15ca8f7090ba

closed 2017-03-30 by TallFurryMan