indilib/indi #206: Limited functionality with Canon 20D

zenmetsu, 2017-04-14:
There appears to be a bug with libgphoto, and it looks like the canon_ccd and gphoto_ccd drivers both have the same issue with this camera.

The problem manifests itself as a proper commanding of the camera to capture an image and subsequent download is OK.  Second capture never triggers a download event as libgphoto apparently goes braindead and doesn't see the new image on the camera.

Adding --keep  to gphoto2 commandline prevents the issue from coming up, however this does not help me in the case of indilib.  I'm not positive that the image delete is causing the problem.  Does indi delete images from camera after capture?  If so, is this configurable somehow so I can test a workaround?

I've replicated this problem with the newest indilib on both debian and gentoo systems.  I've had the same problem with intel and ARM (raspberry pi / pine64).  

I understand that this is certainly a problem in libgphoto2 (one that goes back to 2013 apparently), but I'm hoping that since there is a workaround with gphoto2, a similar workaround can be attempted with indi.  

In the case of indi, after the first capture is made, the camera becomes completely unresponsive to commands.  A usb reset is required to restore functionality.  I can provide any logs upon request.

knro, 2017-04-15:
There is a capture target option. By default it is to RAM, and there is option to save to SD card. Try to see if switching from one to another makes any difference. If this is a GPhoto issue, please file a bug against libgphoto2.

zenmetsu, 2017-04-16:
I managed a "fix" by commenting out the delete section of the gphoto driver.  It works as expected now.  I'll try setting the capture target option to SD card.

It is certainly a bug in libgphoto, but it appears that this is a long-standing issue that has not been addressed, but the issue can certainly be circumvented from the indilib side.  I'll go ahead and close this issue once i report back regarding the RAM vs. SD option.

knro, 2017-04-16:
What section exactly?

zenmetsu, 2017-04-16:
3rdparty/indi-gphoto/gphoto_driver.cpp

`    if (gphoto->upload_settings == GP_UPLOAD_CLIENT)
    {
        DEBUGDEVICE(device, INDI::Logger::DBG_DEBUG,"Deleting.");
        //result = gp_camera_file_delete(gphoto->camera, fn->folder, fn->name, gphoto->context);
        //DEBUGFDEVICE(device, INDI::Logger::DBG_DEBUG,"  Retval: %d", result);
}`

knro, 2017-05-17:
Ok I just pushed a temporary hack to essentially skip the delete step as you did above. Can you verify it works OK with your camera?

closed 2017-10-07 by knro