indilib/indi #228: indi-gpsd has improper DST offset

xsnrg, 2017-05-17:
Discussion: http://www.indilib.org/forum/general/2004-help-with-gpsd.html?start=12#16787

timedatectl returns the proper system settings and DST offset.  indi-gpsd performs a check against the current timezone and changes the offset when it does not need to.  This might be needed on other systems?  On Linux it is handled and the math should not be performed.

    // adjust offset for DST
    if (local->tm_isdst)
        offset -= 1;

credit to:jpaana

knro, 2017-05-17:
@jochym What do you think?

jochym, 2017-05-17:
He is right. We are correcting for DST twice. (!) 
I'll prepare PR in few moments.

xsnrg, 2017-05-18:
Thank you @knro and @jochym

xsnrg, 2017-05-19:
Confirmed fixed with #229

closed 2017-05-19 by xsnrg