indilib/indi #234: FLI ML16200 not detected

gschwim, 2017-05-19:
Reference similar result for CFW-2-7 filter wheel.
[https://github.com/indilib/indi/issues/233](url)

I'm attempting to connect to the FLI ML16200 camera but get an error. I'm using Ekos on Ubuntu w/ the latest builds of everything as of this moment. Errors w/ debug mode on:

```
2017-05-19T07:32:48: Error: no cameras were detected. 
2017-05-19T07:32:48: In find Camera, the domain is 2 
2017-05-19T07:32:48: Attempting to find FLI CCD... 
```

The camera is visible to the OS. I'm able to start TheSkyX and connect to the camera with no problem but indi doesn't see it.

```
Bus 003 Device 006: ID 0f18:000a Finger Lakes Instrumentation ProLine CCD
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x0f18 Finger Lakes Instrumentation
  idProduct          0x000a ProLine CCD
  bcdDevice            1.24
  iManufacturer           1 Finger Lakes Instrumentation, LLC
  iProduct                2 MicroLine ML16200
  iSerial                 3 ML0590516
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered

```

knro, 2017-05-19:
I added more debug statements to the driver. In tomorrow's PPA update, please first open a console window to monitor syslog:

tail -f /var/log/syslog

Then start Ekos/INDI as usual, but before you connect go to INDI FLI tab in the INDI COntrol panel --> Options --> Enable Debug, and check all boxes for Verbose..etc, then try connecting. See what syslog spits out in addition to INDI log as it might help identify the cause of this issue.

gschwim, 2017-05-19:
I will try this soon. In the meanwhile I have new information.

I decided to load the TSX-provided fliusb kernel module and try using Ekos/indi. I'm able to connect to both the filter wheel and the camera. Interesting. Looks like the source is the same so not surprising I guess. 

Digging further, I see the fliusb-dkms package installed. However, the module is not showing up in lsmod. If I run "modprobe fliusb" I get the following:

```
modprobe: ERROR: could not insert 'fliusb': Exec format error
```
Trying with insmod gives me:
```
insmod: ERROR: could not insert module /lib/modules/4.4.0-57-generic/kernel/fliusb.ko: Invalid module format
```

I tried dpkg-reconfigure fliusb-dkms. No help.

```
Error! Module version 1.3 for fliusb.ko
is not newer than what is already found in kernel 4.4.0-57-generic (1.3).
You may override by specifying --force.
```

I've verified that secureboot is disabled.

knro, 2017-05-19:
Thanks, there was indeed some recent changes to the FLI DKMS module to make it compile on more recent kernels. So maybe this broke it under 4.4

@rumengb Can you please take a look at this?

rumengb, 2017-05-19:
Jasem,

Just tried the most recent DKMS on 4.4.0-78-generic and it works. maybe the issue is somewhere else. I will try to figure it out...

rumengb, 2017-05-19:
@gschwim, I think Iknow where is the problem.

can you try something for me:

1. start indiserver by hand as root:
$ sudo indiserver -v indi_fli_ccd

2. Connect to this server using Ekos...

3. Report if it is detected or not

knro, 2017-05-19:
Rumen,

Yes root might help in this case. Permission issue is always second to power issues. We do not have a udev rules for FLI, maybe we should create one for it. Can you control your FLI equipment without root?

gschwim, 2017-05-19:
I tried starting indiserver as root. Didn't help.

If I install the fliusb module provided with TSX complied for my kernel (4.4.0-57-generic) using 'insmod fliusb.ko' things seem to work. 

I also tried starting kstars as root but no luck there.

rumengb, 2017-05-19:
i do not understand. thing seem to work with what software? Ekos?

gschwim, 2017-05-19:
Sorry! I was unclear. Yes, if I load the kernel module provided with TSX, the indiserver seems to be able to use it. I can connect to both the camera and the filterwheel using it. It creates devices /dev/fliusb0 and /dev/fliusb1. Starting indiserver with Ekos allows me to connect to both the camera and filter wheel.

Not sure why I can load that module but not the dkms module.

rumengb, 2017-05-19:
you said dpkg-reconfigure fliusb-dkms returns this:

Error! Module version 1.3 for fliusb.ko
is not newer than what is already found in kernel 4.4.0-57-generic (1.3).
You may override by specifying --force.

can you try "sudo rmmod fliusb" and try again?

gschwim, 2017-05-19:
Yep - same result. No module loads and no devices created.

rumengb, 2017-05-19:
In this case, where can i send you a tarball so that you can try it?

gschwim, 2017-05-19:
Schwim .... Bitrail dot com

gschwim, 2017-05-20:
Rumen - Thanks for the help. Findings for documentation purposes are below.

I built the module you sent and insmod-ed it. It worked fine. This confirms your point that the TSX install was causing the problem. Upon investigating deeper to find out how, I found the TSX install script had this line:

sudo cp ./installfiles/fliusb.ko /lib/modules/$(uname -r)/kernel

The install script has this in spite of it not knowing whether the kernel matches or not. The module it copies over is built on a somewhat older version of the kernel, although the documentation suggests recompiling the module. I removed the fliusb.ko file that it installed completely and the dkms version now works. Interestingly, TSX works with the dkms  version as well, so their install doesn't actually require what they do above. Their linux build of TSX is still very new - I'm one of a few really even testing it as far as I can tell. I'll provide feedback to them shortly. My goal is to just let TSX control the mount w/ TPoint models and let Ekos do all the imaging work. So far things are progressing nicely in that direction.

Thanks much for your help and all the work you do on indi.

closed 2017-05-20 by gschwim