indilib/indi #235: Paramount driver - Ekos Crash

gschwim, 2017-05-19:
I updated my system tonight (apt-get update && apt-get upgrade). What was working before is no longer working. Shortly after connecting to the TCP server of TSX Ekos crashes. If I do not connect to the mount Ekos does not crash. Debug log below for the driver - it doesn't look interesting.

```
INFO	16.064648 sec	: Session log file /home/schwim/.indi/logs/2017-05-19/indi_paramount_telescope/indi_paramount_telescope_07:45:51.log
DEBUG	18.896158 sec	: Toggle Logging Level -- Driver Debug
DEBUG	19.679526 sec	: Toggle Debug Level -- Driver Debug
DEBUG	23.969710 sec	: Toggle Debug Level -- Scope Verbose
DEBUG	24.830992 sec	: Toggle Logging Level -- Scope Verbose
INFO	76.896671 sec	: Connecting to 127.0.0.1@3040 ...
DEBUG	76.896788 sec	: Connection successful, attempting handshake...
DEBUG	76.896813 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.ConnectAndDoNotUnpark();Out = sky6RASCOMTele.IsConnected;
DEBUG	79.933146 sec	: RES: 1|No error. Error = 0.
INFO	79.933308 sec	: Paramount is online.
DEBUG	79.934082 sec	: CMD: /* Java Script */var Out;Out = sky6RASCOMTele.IsTracking;
DEBUG	82.962158 sec	: RES: 0|No error. Error = 0. Paramount.
INFO	82.963531 sec	: Mount is unparked.
DEBUG	82.963731 sec	: CMD: /* Java Script */var Out;Out = sky6RASCOMTele.IsParked();
DEBUG	85.967300 sec	: RES: false|No error. Error = 0.
INFO	85.967405 sec	: Mount is unparked.
DEBUG	85.967511 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	88.970811 sec	: RES: 13.645936804462972,-0.013334417197765205|No error. Error = 0.
SCOPE	88.970923 sec	: Current RA: 13:38:45 Current DEC: -0:00:48
ERROR	88.970968 sec	: Error loading user configuration. Unable to open config file. Error loading file /home/schwim/.indi/Paramount_config.xml: No such file or directory
. To save user configuration, click Save under the Configuration property in the Options tab. 
DEBUG	89.971189 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	92.973936 sec	: RES: 13.647044793450215,-0.013334483642811187|No error. Error = 0.
SCOPE	92.974105 sec	: Current RA: 13:38:49 Current DEC: -0:00:48
DEBUG	93.975025 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	97.041365 sec	: RES: 13.6481888003175,-0.013334442209247148|No error. Error = 0.\ufffd\ufffd
SCOPE	97.041467 sec	: Current RA: 13:38:53 Current DEC: -0:00:48
DEBUG	98.042532 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	101.046197 sec	: RES: 13.649296799694982,-0.013334445965180174|No error. Error = 0.
SCOPE	101.046302 sec	: Current RA: 13:38:57 Current DEC: -0:00:48
DEBUG	102.048211 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	105.049702 sec	: RES: 13.65041681473058,-0.013334355249005884|No error. Error = 0.\ufffd
SCOPE	105.049814 sec	: Current RA: 13:39:02 Current DEC: -0:00:48
DEBUG	106.050872 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	109.054174 sec	: RES: 13.651524812032031,-0.013334371530434309|No error. Error = 0.
SCOPE	109.054302 sec	: Current RA: 13:39:05 Current DEC: -0:00:48
DEBUG	110.055367 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	113.064260 sec	: RES: 13.652632809328543,-0.013334387841683617|No error. Error = 0.
SCOPE	113.064432 sec	: Current RA: 13:39:09 Current DEC: -0:00:48
DEBUG	114.064674 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	117.068262 sec	: RES: 13.653764804202732,-0.013334418767900135|No error. Error = 0.
SCOPE	117.068362 sec	: Current RA: 13:39:14 Current DEC: -0:00:48
DEBUG	118.068554 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	121.069126 sec	: RES: 13.654872814462335,-0.013334356867434359|No error. Error = 0.
SCOPE	121.069385 sec	: Current RA: 13:39:18 Current DEC: -0:00:48
DEBUG	122.070438 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	125.073793 sec	: RES: 13.655984813519742,-0.013334362554479381|No error. Error = 0.
SCOPE	125.073900 sec	: Current RA: 13:39:22 Current DEC: -0:00:48
DEBUG	126.074949 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	129.078310 sec	: RES: 13.65710881784013,-0.013334336487892921|No error. Error = 0.\ufffd
SCOPE	129.078546 sec	: Current RA: 13:39:26 Current DEC: -0:00:48
DEBUG	130.079617 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	133.094972 sec	: RES: 13.65822882040456,-0.013334321015763019|No error. Error = 0.\ufffd
SCOPE	133.095075 sec	: Current RA: 13:39:30 Current DEC: -0:00:48
DEBUG	134.095933 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	137.099267 sec	: RES: 13.659340819461985,-0.013334326702655375|No error. Error = 0.
SCOPE	137.099382 sec	: Current RA: 13:39:34 Current DEC: -0:00:48
DEBUG	138.100447 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	141.103730 sec	: RES: 13.66044481500255,-0.013334353608119515|No error. Error = 0.\ufffd
SCOPE	141.103852 sec	: Current RA: 13:39:38 Current DEC: -0:00:48
DEBUG	142.104230 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	145.161793 sec	: RES: 13.661572797361629,-0.013334460043393784|No error. Error = 0.
SCOPE	145.161905 sec	: Current RA: 13:39:42 Current DEC: -0:00:48
DEBUG	146.162943 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	149.169490 sec	: RES: 13.662700827160132,-0.013334280257058632|No error. Error = 0.
SCOPE	149.169593 sec	: Current RA: 13:39:46 Current DEC: -0:00:48
DEBUG	150.170674 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	153.221116 sec	: RES: 13.663828833241324,-0.013334243567288845|No error. Error = 0.
SCOPE	153.221218 sec	: Current RA: 13:39:50 Current DEC: -0:00:48
DEBUG	154.221652 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	157.224947 sec	: RES: 13.664940832293842,-0.013334249283734913|No error. Error = 0.
SCOPE	157.225051 sec	: Current RA: 13:39:54 Current DEC: -0:00:48
DEBUG	158.226100 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	161.229451 sec	: RES: 13.666048829718303,-0.013334264822767786|No error. Error = 0.
SCOPE	161.229552 sec	: Current RA: 13:39:58 Current DEC: -0:00:48
DEBUG	162.230609 sec	: CMD: /* Java Script */var Out;sky6RASCOMTele.GetRaDec();Out = String(sky6RASCOMTele.dRa) + ',' + String(sky6RASCOMTele.dDec);
DEBUG	165.233886 sec	: RES: 13.667176835671595,-0.013334228904734171|No error. Error = 0.
SCOPE	165.233989 sec	: Current RA: 13:40:02 Current DEC: -0:00:48
```

gschwim, 2017-05-19:
Here's a log of kstars/ekos. This seems to point somewhere:

```

Ekos:  3  devices connected out of  5
WARNING: Application calling GLX 1.3 function "glXCreatePbuffer" when GLX 1.3 is not supported!  This is an application bug!
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:1:1: module "QtQuick" is not installed 
     import QtQuick 2.5 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:2:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 1.4 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:3:1: module "QtQuick.Layouts" is not installed 
     import QtQuick.Layouts 1.1 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:4:1: module "QtQuick.Controls.Styles" is not installed 
     import QtQuick.Controls.Styles 1.4 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:1:1: module "QtQuick" is not installed 
     import QtQuick 2.5 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:2:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 1.4 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:3:1: module "QtQuick.Layouts" is not installed 
     import QtQuick.Layouts 1.1 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:4:1: module "QtQuick.Controls.Styles" is not installed 
     import QtQuick.Controls.Styles 1.4 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:1:1: module "QtQuick" is not installed 
     import QtQuick 2.5 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:2:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 1.4 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:3:1: module "QtQuick.Layouts" is not installed 
     import QtQuick.Layouts 1.1 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:4:1: module "QtQuick.Controls.Styles" is not installed 
     import QtQuick.Controls.Styles 1.4 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:1:1: module "QtQuick" is not installed 
     import QtQuick 2.5 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:2:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 1.4 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:3:1: module "QtQuick.Layouts" is not installed 
     import QtQuick.Layouts 1.1 
     ^
file:///usr/share/kstars/ekos/mount/qml/mountbox.qml:4:1: module "QtQuick.Controls.Styles" is not installed 
     import QtQuick.Controls.Styles 1.4 
     ^
Segmentation fault

```

knro, 2017-05-19:
Please update your KStars, there is a QML package missing. 
sudo apt-get install qml-module-qtquick-controls

gschwim, 2017-05-19:
I manually installed the above. Seems to work now. Thanks!

closed 2017-05-22 by gschwim