indilib/indi #238: Problem with libqhy build in i386 arch

jochym, 2017-05-21:
It seems that the 32 build fails due to the wrong architecture of the blob part of the libqhy (see https://launchpadlibrarian.net/320571332/buildlog_ubuntu-artful-i386.libqhy_0.1.8+3091~ubuntu17.10.1_BUILDING.txt.gz)
I know too little about the qhy build to debug it further bu it seems to be looking for 64bit symbols and libraries:
`cannot find library libc.so.6 needed by debian/libqhy/usr/lib/libqhy.so.0.1.8 (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')`
The failure is universal across all ubuntu releases but is limited to i386 architecture (see https://code.launchpad.net/~jochym/+recipe/libqhy-daily ).

lupinix, 2017-05-21:
I recognized the same on Fedora, across all releases and on i386 only.

knro, 2017-05-21:
I will try to recompile it again today. Thanks for the heads up.

knro, 2017-07-11:
Any issue with QHY still or should I close this? All libraries compile fine now on Launchpad.

jochym, 2017-07-11:
I'll check. Give me some hours.

knro, 2017-07-24:
Please reopen if necessary.

closed 2017-07-24 by knro