indilib/indi #242: indi_v4l2_ccd crash

jakugab, 2017-05-25:
Hello!
I have a Raspberry Pi 2 running a freshly installed Ubuntu MATE 16.04 for the RPi (firmware version: 4.1.19 v7+). I attached the dedicated RPi NoIR V2 camera board, which works fine with the camera app (I can take photos with raspistill command from terminal), and streaming program, like Cheese (using bcm2835_v4l2 module). I installed the latest indi-full and kstars-bleeding from the ppa successfully. In Ekos I made a profile with only a CCD (I chose V4L2 CCD from CCD tab). When I start INDI, the ccd is trying to connect, but fails, Kstars says: "Kstars detected, that indi_v4l2_ ccd driver crashed, please check server log in the Device Manager ". Here is the Kstars-log of the latest crash:

 2017-05-17T09:30:53.652 - DEBG - Welcome to KStars 2.7.7
2017-05-17T09:30:53.677 - DEBG - "Setting clock:  UTC: Sze máj. 17 07:30:53 2017 GMT  JD: 2Â 457Â 890,81"
2017-05-17T09:30:54.135 - DEBG - Opened the DSO Database. Ready!
2017-05-17T09:30:54.637 - DEBG - Opened the User DB. Ready.
2017-05-17T09:30:59.893 - DEBG - Processing  "unnamedstars.dat" , HTMesh Level 3
2017-05-17T09:30:59.894 - DEBG -   Sky Mesh Size:  512
2017-05-17T09:31:00.074 - DEBG - Loaded DSO catalog file:  "unnamedstars.dat"
2017-05-17T09:31:00.081 - DEBG - Processing  "deepstars.dat" , HTMesh Level 3
2017-05-17T09:31:00.082 - DEBG -   Sky Mesh Size:  512
2017-05-17T09:31:00.082 - DEBG - Loaded DSO catalog file:  "deepstars.dat"
2017-05-17T09:31:02.366 - WARN - "Star HD61421 not found."
2017-05-17T09:31:02.382 - WARN - "Star HD10700 not found."
2017-05-17T09:31:02.409 - WARN - "Star HD20794 not found."
2017-05-17T09:31:02.435 - WARN - "Star HD2151 not found."
2017-05-17T09:31:02.468 - WARN - "Star HD190248 not found."
2017-05-17T09:31:02.505 - WARN - "Star HD142860 not found."
2017-05-17T09:31:02.515 - WARN - "Star HD1581 not found."
2017-05-17T09:31:02.516 - WARN - "Star HD1581 not found."
2017-05-17T09:31:02.521 - WARN - "Star HD98230 not found."
2017-05-17T09:31:02.624 - DEBG - File opened:  "/home/ubuntu/.local/share/kstars/ngcic.dat"
2017-05-17T09:31:02.624 - DEBG - Loading NGC/IC objects
2017-05-17T09:31:07.852 - DEBG - File opened:  "/usr/share/kstars/asteroids.dat"
2017-05-17T09:31:09.155 - DEBG - File opened:  "/usr/share/kstars/comets.dat"
2017-05-17T09:31:11.828 - DEBG - No WishList Saved yet
2017-05-17T09:31:16.346 - WARN - "Object named NGC 235 not found"
2017-05-17T09:31:16.465 - DEBG - Starting the timer
2017-05-17T09:31:16.466 - DEBG - Daylight Saving Time active
2017-05-17T09:31:16.466 - DEBG - Next Daylight Savings Time change (Local Time):  "V okt. 29 03:00:00 2017 GMT"
2017-05-17T09:31:16.467 - DEBG - Next Daylight Savings Time change (UTC):  "V okt. 29 00:59:59 2017 GMT"
2017-05-17T09:31:17.417 - DEBG - The current Date/Time is:  "Sze máj. 17 09:31:17 2017 GMT+0200"
2017-05-17T09:31:17.417 - DEBG - glibc >= 2.1 detected.  Using GNU extension sincos()
2017-05-17T09:31:19.231 - WARN - "Object named NGC 526 not found"
2017-05-17T09:31:21.933 - WARN - "Object named NGC 651 not found"
2017-05-17T09:31:25.978 - WARN - "Object named NGC 3314 not found"
2017-05-17T09:31:28.762 - WARN - "Object named NGC 3690 not found"
2017-05-17T09:31:32.368 - WARN - "Object named NGC 6027 not found"
2017-05-17T09:31:36.198 - WARN - "Object named IC 1179 not found"
2017-05-17T09:31:39.049 - WARN - "Object named Europa not found"
2017-05-17T09:31:49.949 - DEBG - Ekos:  "Starting INDI services..."
2017-05-17T09:31:49.951 - DEBG - INDI: Starting local drivers...
2017-05-17T09:31:49.996 - DEBG - INDI: INDI Server started locally on port  7624
2017-05-17T09:31:49.998 - DEBG - INDIListener: Adding a new client manager to INDI listener..
2017-05-17T09:31:49.999 - DEBG - INDI: Connecting to local INDI server on port  7624  ...
2017-05-17T09:31:50.003 - DEBG - Connection to INDI server is successful
2017-05-17T09:31:50.004 - DEBG - Ekos:  "INDI services started on port 7624."
2017-05-17T09:31:50.058 - DEBG - Received new device  V4L2 CCD
2017-05-17T09:31:50.070 - DEBG - INDI Server:  "2017-05-17T07:31:49: startup: /usr/bin/indiserver -v -p 7624 -m 100 -f /tmp/indififo2e6fa1e6 "
2017-05-17T09:31:50.071 - DEBG - INDI Server:  "2017-05-17T07:31:50: listening to port 7624 on fd 3"
2017-05-17T09:31:50.071 - DEBG - INDI Server:  "FIFO: start indi_v4l2_ccd -n \"V4L2 CCD\""
2017-05-17T09:31:50.071 - DEBG - INDI Server:  "With name: V4L2 CCD"
2017-05-17T09:31:50.071 - DEBG - INDI Server:  "FIFO: Starting driver indi_v4l2_ccd"
2017-05-17T09:31:50.072 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: pid=2714 rfd=4 wfd=7 efd=8"
2017-05-17T09:31:50.072 - DEBG - INDI Server:  "2017-05-17T07:31:50: Client 5: new arrival from 127.0.0.1:52792 - welcome!"
2017-05-17T09:31:50.072 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: Using default decoder 'Builtin decoder'"
2017-05-17T09:31:50.073 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd:   Supported V4L2 formats are:"
2017-05-17T09:31:50.073 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: Y16  "
2017-05-17T09:31:50.073 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: NV21 "
2017-05-17T09:31:50.073 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: BA81 "
2017-05-17T09:31:50.073 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: YU12 "
2017-05-17T09:31:50.074 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: NV12 "
2017-05-17T09:31:50.074 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: YV12 "
2017-05-17T09:31:50.074 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: BYR2 "
2017-05-17T09:31:50.074 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: RGB3 "
2017-05-17T09:31:50.074 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: RGGB "
2017-05-17T09:31:50.075 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: JPEG "
2017-05-17T09:31:50.075 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: MJPG "
2017-05-17T09:31:50.075 - DEBG - INDI Server:  "2017-05-17T07:31:50: Driver indi_v4l2_ccd: RGBO "
[... 36 lines cut ...]
2017-05-17T09:31:52.576 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: RGGB "
2017-05-17T09:31:52.576 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: JPEG "
2017-05-17T09:31:52.576 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: MJPG "
2017-05-17T09:31:52.576 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: RGBO "
2017-05-17T09:31:52.577 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: RGBP "
2017-05-17T09:31:52.577 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: YVYU "
2017-05-17T09:31:52.577 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: YUYV "
2017-05-17T09:31:52.578 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: GREY "
2017-05-17T09:31:52.578 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: VYUY "
2017-05-17T09:31:52.578 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: UYVY "
2017-05-17T09:31:52.578 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: snooping
 on Telescope Simulator.EQUATORIAL_EOD_COORD"
2017-05-17T09:31:52.579 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: snooping on Telescope Simulator.TELESCOPE_INFO"
2017-05-17T09:31:52.579 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: snooping on CCD Simulator.FILTER_SLOT"
2017-05-17T09:31:52.579 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: snooping on CCD Simulator.FILTER_NAME"
2017-05-17T09:31:52.580 - DEBG - INDI Server:  "2017-05-17T07:31:52: Driver indi_v4l2_ccd: snooping on SQM.SKY_QUALITY"
2017-05-17T09:31:52.580 - DEBG - INDI Server:  ""
2017-05-17T09:32:19.632 - DEBG - Reporting new timestep value:  60

After the crash, sometimes, but not in all case, Ubuntu sends an error message too, I attached the ".crash" file in a supported format. 
I updated the firmware (sudo rpi-update), than I tried to remove and reinstall kstars-bleeding and indi-full without success, I always got this error. I tried to upgrade the obsolate packages, that Ubuntu error message mentioned, but the indi_v4l2_ ccd crashed again. I dont't know is it a bug, or something else, maybe a problem with my system, but it seems to a consequent error.

P.S.: I am just an end-user, I' m very new in Linux, Kstars/INDI, and I have no programming skills at all :-(
Thanks,
Gabor

TallFurryMan, 2017-05-28:
Hello, there seems to be a string formatting going wrong while the driver checks the device capabilities. I'm not able to have a look right now, it's in the big "check_device" function which gathers characteristics by issuing ioctls. If no one else picks the point, I'll check this mid to end this week.

jakugab, 2017-05-29:
Thank you in advance for your help, I'm waiting patiently for the result.

TallFurryMan, 2017-06-04:
@jakugab please test your scenario with the update from the ppa (in ~12h) now that #248 is merged. Please provide the same info as you already did, that was perfect.

knro, 2017-06-04:
Looks like I merged prematurely, can you please fix these warning?

Also, why are we writing to stderr? Use INDI::Logger whenever possible.

[ 35%] Building CXX object CMakeFiles/indidriverstatic.dir/libs/webcam/v4l2_base.cpp.o
/movies/Dropbox/Projects/indi/libindi/libs/webcam/v4l2_base.cpp: In member function ‘int V4L2_Base::init_mmap(char*)’:
/movies/Dropbox/Projects/indi/libindi/libs/webcam/v4l2_base.cpp:845:98: warning: field precision specifier ‘.*’ expects argument of type ‘int’, but argument 3 has type ‘long unsigned int’ [-Wformat=]
             fprintf (stderr, "%.*s does not support memory mapping\n", sizeof(dev_name), dev_name);

TallFurryMan, 2017-06-04:
I did not get this warning on my 32-bit machine. In any case we should wait for @jakugab's report before closing.

jakugab, 2017-06-04:
First of all, sorry for the late answer. I followed all of your work to fix the problem, thanks a lot. So I reinstalled indi-full and kstars-bleeding from ppa just now, but when I start Ekos, indi_v4l2_ ccd driver crashes again with the same manner and same error message, as earlier. :-(

TallFurryMan, 2017-06-05:
Same crash report? If I read the ppa correctly, version embedding the first step of the fix is 1.4.1+r3120~201706041724~ubuntu17.10.1, could you confirm your test environment?

jakugab, 2017-06-05:
The installed packeges from the ppa are: kstars-bleeding (5:17.04+201706041301~ubuntu16.04.1), indi-full (1.4~201706041718~ubuntu16.04.1), libindi (libindi1_1.4.1+r3120~201706041724~ubuntu16.04.1). I'm running MATE 16.04 for the RPi with Rpi camera board.
I attached INDI log (indi_v4l2_ccd_23:02:10_log.txt), Kstars log, and Ubuntu crash report (_usr_bin_indi_v4l2_ccd_crash.txt) about the latest crash. What am I doing wrong?

[Kstars log_18-23-44.txt](https://github.com/indilib/indi/files/1052751/Kstars.log_18-23-44.txt)
[indi_v4l2_ccd_23:02:10_ log.txt](https://github.com/indilib/indi/files/1052746/indi_v4l2_ccd_23.02.10_.log.txt)
[_usr_bin_indi_v4l2_ccd_crash.txt](https://github.com/indilib/indi/files/1052755/_usr_bin_indi_v4l2_ccd_crash.txt)

jakugab, 2017-06-05:
Sorry for the crossed words, this was accidental.

TallFurryMan, 2017-06-05:
No issue, that's perfectly readable. I'll have a look to verify, and will probably have to ask you to debug the driver to provide a backtrace.

TallFurryMan, 2017-06-05:
From what I understand from the ppa information, the version I mentioned does not contain the first pull request I pushed yesterday. @knro could you confirm?

TallFurryMan, 2017-06-05:
Sorry r3120 indeed contains my first pull request. I'll check the code again, @jakugab this will probably require you to execute the debugger with "gdb --eval-command="set follow-fork-mode child" --args indiserver -v indi_v4l2_ccd" to provide a backtrace, although I'm not sure we will get more information without rebuilding the driver in debug.

TallFurryMan, 2017-06-05:
Debug symbols are embedded in package "indi-dbg-1.4.1+3120~ubuntu16.04.1".

TallFurryMan, 2017-06-05:
Ok found it. An exquisitely misplaced parenthesis :)

TallFurryMan, 2017-06-06:
@jakugab r3121 should fix your issue. Sorry it went under the radar through my tests like this.

jakugab, 2017-06-06:
Thank you very much, I will try to update today, and I'll report the result.

jakugab, 2017-06-06:
I updated to r3121 all of the drivers and libindi. Kstars-bleeding is the latest version. When I start Ekos, V4L2 CCD is now connecting successfully :-), I can reach all the options in the INDI Control Panel, but when I want to make a capture in the Capture module, the driver crashes, and the capturing sequence doesn't stop. I attached the log files. When I restart INDI, the driver's connecting fine, but Ekos initialises the camera with different resolutions (once 1024x768, once 3280x2464 -this is the resolution of the RPi cam, and 32x32), and I don't know why?

[INDI Control panel log.txt](https://github.com/indilib/indi/files/1056441/INDI.Control.panel.log.txt)
[Kstars Device Manager log.txt](https://github.com/indilib/indi/files/1056446/Kstars.Device.Manager.log.txt)
[Kstars log_21-12-25.txt](https://github.com/indilib/indi/files/1056447/Kstars.log_21-12-25.txt)

TallFurryMan, 2017-06-07:
Wow, how come this use case is so crippled...? Could you enable debug in the V4L2 driver panel options, check debug logs box, retry your scenario and attach the results?

TallFurryMan, 2017-06-07:
Sorry you did that already. I was looking for "/home/ubuntu/.indi/logs/2017-06-06/indi_v4l2_ccd/indi_v4l2_ccd_18:58:51.log"

TallFurryMan, 2017-06-07:
With debug enabled, when I run an exposure directly from the INDI V4L2 CCD panel, tab Main Control, this is what I get in the console at the bottom of the window:
```
2017-06-07T11:59:59: Upload complete 
2017-06-07T11:59:58: Uploading file. Ext: fits, Size: 311040, sendImage? Yes, saveImage? No 
2017-06-07T11:59:58: Capture of one frame (0 stacked frames) took 10.032493 seconds. 
2017-06-07T11:59:58: Stopping running exposure 1.000 seconds before completion 
2017-06-07T11:59:58: read_frame: [0x9afd2e8] decoding 307200-byte buffer 0xb6bcd000 cropset N 
2017-06-07T11:59:58: read_frame: frame exposure started 0.031 seconds ago 
2017-06-07T11:59:58: is_compressed: format GREY has compressed flag 0 
2017-06-07T11:59:58: read_frame: buffer #0 dequeued from fd:3 
2017-06-07T11:59:58: read_frame: using MMAP to recover frame buffer 
2017-06-07T11:59:48: Using device 100000-tick manual exposure for 10.000-second exposure 
```
I didn't see that sort of log in your report. Could you provide the contents of that console when you connect and use the V4L2 driver?

jakugab, 2017-06-07:
Sorry, I forgot to attach the requested INDI log file, now I attach it, but that's an empty file.
[indi_v4l2_ccd_18:58:51.txt](https://github.com/indilib/indi/files/1059164/indi_v4l2_ccd_18.58.51.txt)
When I start INDI, the V4L2 CCD connects, and in the Option tab debug enabled, the content of the log window is:

2017-06-07T19:32:18: Session log file /home/ubuntu/.indi/logs/2017-06-07/indi_v4l2_ccd/indi_v4l2_ccd_19:32:00.log 
2017-06-07T19:32:16: Toggle Logging Level -- Driver Debug 
2017-06-07T19:32:15: Toggle Debug Level -- Driver Debug 
2017-06-07T19:32:09: Debug is enabled. 
2017-06-07T19:32:01: Device configuration applied. 
2017-06-07T19:32:01: Upload settings set to client only. 
2017-06-07T19:32:01: Loading device configuration... 
2017-06-07T19:32:01: Found intial Input "Camera 0", Format "Planar YUV 4:2:0", Size 3280x2464, Frame interval 1/30s 
2017-06-07T19:32:01: V4L2 CCD Device is online. Initializing properties. 

When I start an exposure in the Main Control tab, I get error message "Kstars detected, INDI driver indi_v4l2_ccd crashed...", and in the log window appears just one additional line:
2017-06-07T19:34:32: Using default recorder (SER File Recorder). 
See Screenshot1 in the attachments. I attached screenshots from the relevant tabs of the V4L2 CCD panel, Kstars and INDI logs too. I didn't make any settings, just ran an exposure in the Main Control tab.
[indi_v4l2_ccd_19:32:00.log.txt](https://github.com/indilib/indi/files/1059208/indi_v4l2_ccd_19.32.00.log.txt)
[log_21-30-53.txt](https://github.com/indilib/indi/files/1059207/log_21-30-53.txt)
[image]
[image]
[image]
[image]
[image]
[image]

TallFurryMan, 2017-06-07:
I see the default capture format is JFIF. Would you mind trying with GREY for instance? Then try out a few formats, to see which works and which doesn't.

jakugab, 2017-06-08:
Sorry for the late answer. Today, when I started the process and V4L2 CCD connects, I got a different window log, then earlier, similar as yours. I tried all of the capture formats in V4L2 Capture Options tab but the driver always crashed. I didn't find GREY format here, but I see it in the Kstars log as a supported format. Would you please help me, where can I search it? Thanks.

[V4L2 CCD window log.txt](https://github.com/indilib/indi/files/1060981/V4L2.CCD.window.log.txt)
[indi_v4l2_ccd_12:15:41 log.txt](https://github.com/indilib/indi/files/1060984/indi_v4l2_ccd_12.15.41.log.txt)
[log_14-14-25.txt](https://github.com/indilib/indi/files/1060985/log_14-14-25.txt)

TallFurryMan, 2017-06-08:
Driver says current format is JPEG, so I'd assume RGB, and also says it is updating the frame size to 786432 bytes.
A RGB frame sized to 1024x768 is 2359296 bytes max, so the only input I see which could make the driver choose 786432 bytes is the "Gray" value of the button Image Type in the Image Settings.
Could you try switching Image Type to "Color" to see if it has an effect?

In the previous reports you had a crash log named `_usr_bin_indi_v4l2_ccd_crash.txt`, that was interesting if you can provide this too because it contains a backtrace of the crash. Log `indi_v4l2_ccd_HH:MM:SS log.txt` is interesting too as it shows debug traces from the driver. The two others logs more or less duplicate the information, so you can probably ignore them.

jakugab, 2017-06-09:
I changed to "Color", ran an exposure and the driver chrashed. Here are the log files:
[_usr_bin_indi_v4l2_ccd_crash.txt](https://github.com/indilib/indi/files/1063910/_usr_bin_indi_v4l2_ccd_crash.txt)

[indi_v4l2_ccd_11:24:16 log.txt](https://github.com/indilib/indi/files/1063920/indi_v4l2_ccd_11.24.16.log.txt)
[V4L2CCD window log.txt](https://github.com/indilib/indi/files/1063946/V4L2CCD.window.log.txt)

TallFurryMan, 2017-06-09:
Thanks for those logs! It seems it is the AbsExposureN pointer which is not managed properly. I'll try to push a fix for this, and have a look around to understand the cause and consequences. Stay tuned.

jakugab, 2017-06-09:
I'm happy to hear that, thanks in advance!

jakugab, 2017-06-11:
I made an update. I started Ekos, V4L2 CCD connected, and if I ran an expo either from Main Control tab of V4L2 CCD panel in INDI Control Panel, or from the CCD panel in Ekos, the driver **doesn't crashed** (hurray!!!!  :-)), but I got a message: " Failed 1.000-second manual exposure, out of device tick bounds [0,0] " and " No exposure or streaming in progress". I tried it in gray and color too, with the same message.
[indi_v4l2_ccd_09:54:15_ log.txt](https://github.com/indilib/indi/files/1066343/indi_v4l2_ccd_09.54.15_.log.txt)
[V4L2 CCD window log.txt](https://github.com/indilib/indi/files/1066344/V4L2.CCD.window.log.txt)

TallFurryMan, 2017-06-11:
I suggest we close this issue and open a new one, now that the crash is behind us. It doesn't work yet, but maybe someone will have already experienced this with the raspberry camera. I'll continue checking what happened to AbsExposureN in your situation.

TallFurryMan, 2017-06-11:
Maybe the OP http://www.indilib.org/forum/ccds-dslrs/861-raspberry-camera-long-exposure-with-v4l2-driver.html has information. Let's make this work!

jakugab, 2017-06-12:
I agree with closing the issue. I'll check the link, maybe I'll find the solution. I want to say a great thanks for your hard work to solve this issue, and for the further investigation! Solving this issue was a great experience for me, I learned much. Thank you again! 
Best regards
Gabor

closed 2017-06-12 by jakugab

TallFurryMan, 2017-07-14:
@jakugab The follow-up you gave on this issue ("Failed xxx exposure...") is probably fixed by PR #299, if you want to give a try on your setup.

jakugab, 2017-07-17:
Thanks a lot, I'll try it as I'll have a little time, and  I will report the result, of course.

PR #248 (2017-06-04): Secured formatting of strings in static arrays
Secured formatting of strings in static arrays with their own array size.
Let's see if this is enough to fix #242.