indilib/indi #243: indiserver dies when attempting to use indi_gphoto_ccd from kstars to access a Sony A6000

justanenduser, 2017-05-28:
Running Kstars from gdb gives the following, 

Dispatch command error(-1): INDI: Could not find property CCD_EXPOSURE in GPhoto CCD
<setNumberVector device="GPhoto CCD" name="CCD_EXPOSURE" state="Idle" timeout="60" timestamp="2017-05-28T02:07:19">
    <oneNumber name="CCD_EXPOSURE_VALUE" min="0.001" max="3600" step="1">

Kstars, libgphoto2 and libindi have been built from GitHub within the past 24 hours. 
Please advise if I can provide any other info.

Tnx.

knro, 2017-05-28:
Try to debug the INDI Gphoto driver

gdb --eval-command="set follow-fork-mode child" --args indiserver -v indi_gphoto_ccd 

Then connect to it from KStars (update profile to only include GPhoto driver), set mode to remote and keep host as _localhost_ then connect from Ekos.

If it crashes, type **bt** to get the backtrace.

justanenduser, 2017-05-28:
I rebuilt libindi and the gphoto_ccd driver with the debug option.

[gary@bigboote ~]$ gdb --eval-command="set follow-fork-mode child" --args
indiserver -v indi_gphoto_ccd
GNU gdb (GDB) Fedora 7.12.1-48.fc25
Reading symbols from indiserver...done.
(gdb) run
Starting program: /usr/local/bin/indiserver -v indi_gphoto_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
2017-05-28T15:47:40: startup: /usr/local/bin/indiserver -v indi_gphoto_ccd
2017-05-28T15:47:40: Driver indi_gphoto_ccd: pid=14750 rfd=3 wfd=6 efd=7
2017-05-28T15:47:40: listening to port 7624 on fd 4
[New process 14750]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
process 14750 is executing new program: /usr/local/bin/indi_gphoto_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
2017-05-28T15:47:41: Driver indi_gphoto_ccd: snooping on Telescope
Simulator.EQUATORIAL_EOD_COORD
2017-05-28T15:47:41: Driver indi_gphoto_ccd: snooping on Telescope
Simulator.TELESCOPE_INFO
2017-05-28T15:47:41: Driver indi_gphoto_ccd: snooping on CCD
Simulator.FILTER_SLOT
2017-05-28T15:47:41: Driver indi_gphoto_ccd: snooping on CCD
Simulator.FILTER_NAME
2017-05-28T15:47:41: Driver indi_gphoto_ccd: snooping on SQM.SKY_QUALITY
2017-05-28T15:48:35: Client 0: new arrival from 127.0.0.1:45476 - welcome!
[New Thread 0x7ffff2468700 (LWP 14763)]
[Thread 0x7ffff2468700 (LWP 14763) exited]
[New Thread 0x7ffff2468700 (LWP 14764)]
[Thread 0x7ffff2468700 (LWP 14764) exited]
[New Thread 0x7ffff2468700 (LWP 14765)]
[New Thread 0x7ffff1c67700 (LWP 14766)]

Thread 2.1 "indi_gphoto_ccd" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff7f9d8c0 (LWP 14750)]
0x000000000041a224 in find_exposure_setting (gphoto=0x65bd30,
widget=0x67d910, exptime_usec=1000000)
    at
/home/gary/Downloads/indi-master/3rdparty/indi-gphoto/gphoto_driver.cpp:535
535        DEBUGFDEVICE(device, INDI::Logger::DBG_DEBUG,"Best match: %g
seconds Index: %d", gphoto->exposure[best_idx], best_idx);
(gdb) bt full
#0  0x000000000041a224 in find_exposure_setting (gphoto=0x65bd30,
widget=0x67d910, exptime_usec=1000000)
    at
/home/gary/Downloads/indi-master/3rdparty/indi-gphoto/gphoto_driver.cpp:535
        i = 0
        exptime = 1
        best_idx = 0
        delta = 6.9533558070729575e-310
        best_match = 99999
#1  0x000000000041badc in gphoto_start_exposure (gphoto=0x65bd30,
exptime_usec=1000000, mirror_lock=0)
    at
/home/gary/Downloads/indi-master/3rdparty/indi-gphoto/gphoto_driver.cpp:773
        idx = -139953104
#2  0x0000000000412046 in GPhotoCCD::StartExposure (this=0x63c1c0,
duration=1)
    at
/home/gary/Downloads/indi-master/3rdparty/indi-gphoto/gphoto_ccd.cpp:957
        exp_us = 1000000
#3  0x00007ffff7a8ba7c in INDI::CCD::ISNewNumber (this=0x63c1c0,
dev=0x63c100 "GPhoto CCD", name=0x63c030 "CCD_EXPOSURE",
    values=0x6bd720, names=0x6c9c80, n=1) at
/home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp:974
No locals.
#4  0x0000000000410fa2 in GPhotoCCD::ISNewNumber (this=0x63c1c0,
[... 67 lines cut ...]
              prev = 0x0, cleanup = 0x0, canceltype = -8440}}}
        not_first_call = <optimized out>
#13 0x000000000040d8ca in _start ()
No symbol table info available.
(gdb) q
A debugging session is active.

    Inferior 2 [process 14750] will be killed.

Quit anyway? (y or n) y
Child process 14750 died
2017-05-28T15:50:04: Driver indi_gphoto_ccd: stderr EOF
2017-05-28T15:50:04: Driver indi_gphoto_ccd: restart #1
2017-05-28T15:50:04: Driver indi_gphoto_ccd: pid=14772 rfd=3 wfd=7 efd=8
2017-05-28T15:50:04: Driver indi_gphoto_ccd: snooping on Telescope
Simulator.EQUATORIAL_EOD_COORD
2017-05-28T15:50:04: Driver indi_gphoto_ccd: snooping on Telescope
Simulator.TELESCOPE_INFO
2017-05-28T15:50:04: Driver indi_gphoto_ccd: snooping on CCD
Simulator.FILTER_SLOT
2017-05-28T15:50:04: Driver indi_gphoto_ccd: snooping on CCD
Simulator.FILTER_NAME
2017-05-28T15:50:04: Driver indi_gphoto_ccd: snooping on SQM.SKY_QUALITY
[gary@bigboote ~]$ 2017-05-28T15:50:12: Client 0: read EOF
2017-05-28T15:50:12: Client 0: shut down complete - bye!

knro, 2017-05-29:
Thanks, that was helpful. Looks like there is something wrong with your exposure widget from GPhoto. Can you run the driver again, but turn on Debug Logging to a file _before_ you connect (Under Options). Then after you connect, the file should contain some information about the gphoto widgets. Please attach this file.

justanenduser, 2017-05-29:
As requested. Attached is the debug log file.

INFO	44.338205 sec	: Session log file /home/gary/.indi/logs/2017-05-29/indi_gphoto_ccd/indi_gphoto_ccd_14:50:18.log
DEBUG	45.426065 sec	: Toggle Logging Level -- Driver Debug
DEBUG	45.866008 sec	: Toggle Debug Level -- Driver Debug
DEBUG	60.156411 sec	: Mirror lock value: 0.000000
DEBUG	60.156502 sec	: Opening gphoto
DEBUG	60.156587 sec	: Camera init.  Takes about 10 seconds.
DEBUG	60.343620 sec	: Setting toggle widget capture: 1
DEBUG	60.351630 sec	: Setting new configuration OK.
DEBUG	60.351849 sec	: Finding exposure widget...
WARNING	60.351936 sec	: Shutter speed widget does not have any valid data (count=0)
DEBUG	60.352006 sec	: Exposure Widget: shutterspeed
DEBUG	60.352081 sec	: ISO Widget: iso
DEBUG	60.352152 sec	: Finding bulb widget...
DEBUG	60.352219 sec	: Bulb Widget: bulb
DEBUG	60.352277 sec	: 		Value: On
DEBUG	60.352345 sec	: Manufacturer: Sony Corporation
DEBUG	60.352405 sec	: Model: ILCE-6000
DEBUG	60.352460 sec	: Using external shutter release port: /dev/ttyUSB0
DEBUG	60.352518 sec	: GPhoto initialized.
DEBUG	60.352673 sec	: Blub-stop thread enabled.
INFO	60.352885 sec	: GPhoto CCD is online.
INFO	60.352940 sec	: Detected Sony Corporation Model ILCE-6000.
DEBUG	60.374467 sec	: Available options
DEBUG	60.374579 sec	: 	autofocus:
DEBUG	60.374638 sec	: 		Value: On
DEBUG	60.374688 sec	: 	capture:
DEBUG	60.374739 sec	: 		Value: On
DEBUG	60.374790 sec	: 	movie:
DEBUG	60.374840 sec	: 		Value: On
DEBUG	60.374890 sec	: 	opcode:
DEBUG	60.374941 sec	: 		Value: 0x1001,0xparam1,0xparam2
DEBUG	60.374994 sec	: 	serialnumber:
DEBUG	60.375044 sec	: 		Value: 00000000000000003282671003486209
DEBUG	60.375096 sec	: 	manufacturer:
DEBUG	60.375151 sec	: 		Value: Sony Corporation
DEBUG	60.375203 sec	: 	cameramodel:
DEBUG	60.375253 sec	: 		Value: ILCE-6000
DEBUG	60.375303 sec	: 	deviceversion:
DEBUG	60.375353 sec	: 		Value: 3.0
DEBUG	60.375406 sec	: 	vendorextension:
DEBUG	60.375457 sec	: 		Value: Sony PTP Extensions
DEBUG	60.375509 sec	: 	batterylevel:
DEBUG	60.375559 sec	: 		Value: 100%
DEBUG	60.375609 sec	: 	imagesize:
DEBUG	60.375661 sec	: 		    0: Unknown value 0004
DEBUG	60.375712 sec	: 		    1: Unknown value 000e
DEBUG	60.375763 sec	: 		    2: Unknown value 000f
DEBUG	60.375814 sec	: 		    3: Unknown value 0005
DEBUG	60.375865 sec	: 		    4: Unknown value 0006
DEBUG	60.375916 sec	: 		    5: Unknown value 0007
DEBUG	60.375966 sec	: 		    6: Unknown value 0010
DEBUG	60.376017 sec	: 		    7: Unknown value 0011
DEBUG	60.376068 sec	: 		    8: Unknown value 0012
DEBUG	60.376122 sec	: 		    9: Unknown value 0008
DEBUG	60.376177 sec	: 		   10: Unknown value 0013
DEBUG	60.376228 sec	: 		   11: Unknown value 0009
DEBUG	60.376279 sec	: 		   12: Unknown value 000a
DEBUG	60.376330 sec	: 		   13: Unknown value 000b
DEBUG	60.376381 sec	: 		   14: Unknown value 0014
DEBUG	60.376432 sec	: 		   15: Unknown value 0015
DEBUG	60.376483 sec	: 		   16: Unknown value 000c
DEBUG	60.376535 sec	: 		   17: Unknown value 0017
DEBUG	60.376586 sec	: 		   18: Unknown value 0018
DEBUG	60.376636 sec	: 		   19: Unknown value 000d
DEBUG	60.376687 sec	: 		*  20: Large
DEBUG	60.376736 sec	: 		   21: Medium
DEBUG	60.376787 sec	: 		   22: Small
DEBUG	60.376836 sec	: 	colortemperature:
[... 1360 lines cut ...]
DEBUG	60.481634 sec	: 	   44: 1250 Multi Frame Noise Reduction
DEBUG	60.481657 sec	: 	   45: 1600 Multi Frame Noise Reduction
DEBUG	60.481679 sec	: 	   46: 2000 Multi Frame Noise Reduction
DEBUG	60.481701 sec	: 	   47: 2500 Multi Frame Noise Reduction
DEBUG	60.481723 sec	: 	   48: 3200 Multi Frame Noise Reduction
DEBUG	60.481746 sec	: 	   49: 4000 Multi Frame Noise Reduction
DEBUG	60.481768 sec	: 	   50: 5000 Multi Frame Noise Reduction
DEBUG	60.481796 sec	: 	   51: 6400 Multi Frame Noise Reduction
DEBUG	60.481820 sec	: 	   52: 8000 Multi Frame Noise Reduction
DEBUG	60.481842 sec	: 	   53: 10000 Multi Frame Noise Reduction
DEBUG	60.481865 sec	: 	   54: 12800 Multi Frame Noise Reduction
DEBUG	60.481887 sec	: 	   55: 16000 Multi Frame Noise Reduction
DEBUG	60.481909 sec	: 	   56: 25600 Multi Frame Noise Reduction
DEBUG	60.481932 sec	: 	   57: 51200 Multi Frame Noise Reduction
DEBUG	60.487436 sec	: Toggle Debug Level -- Driver Debug
DEBUG	60.487519 sec	: Toggle Logging Level -- Driver Debug
INFO	60.487591 sec	: Session log file /home/gary/.indi/logs/2017-05-29/indi_gphoto_ccd/indi_gphoto_ccd_14:50:18.log
DEBUG	60.487793 sec	: Requested CCD Frame is (  0,  0) (6000 x 4000)
DEBUG	60.488342 sec	: Configuration successfully loaded.
DEBUG	95.774321 sec	: Starting exposure (exptime: 1 secs, mirror lock: 0)
DEBUG	95.774361 sec	:   Mutex locked
DEBUG	95.774371 sec	: Setting radio/menu widget iso: 11 (800)
DEBUG	95.774541 sec	: Setting new configuration OK.
DEBUG	95.774549 sec	: Using camera predefined exposure ranges.
DEBUG	95.774556 sec	: Finding optimal exposure setting for 1 seconds in shutterspeed (count=0)...

knro, 2017-05-30:
So your exposure widget has 0 members, which was causing a crash. Anyway, I fixed the crash and forced it to resort to bulb whenever this is the case. I suspect this is probably due to missing support from GPhoto. I also now included libgphoto2 version information in the log so we know what to deal with. Please test again and let me know how it goes.

justanenduser, 2017-05-30:
Thank you.
indi_gphoto_ccd and my Sony A6000 now talk to each other ... :-)

closed 2017-06-04 by knro