indilib/indi #251: saving the first capture (and only the first one) on Sony A6000 fails

justanenduser, 2017-06-06:
The first  attempt at a capture will fail (whether using CCDciel or Kstars) when the image is being downloaded. If I wait for the camera to clear/reset (which it will do) all subsequent captures or previews (single shot, looping or sequence) work just fine. 

The specific error message received "Exposure failed to save image... Unspecified error" seems to come from gphoto_cdd.cpp line 1197. 

LIBINDI, LIBGPHOTO2 and KSTARS have all been built, within the last week, from their respective GitHub source. Patrick Chevalley has provided an updated CCDciel in his attempts. 

What follows is an extract of the portion of the log that show the exposure and error.

DEBUG	69.608849 sec	: Starting exposure (exptime: 1 secs, mirror lock: 0)
DEBUG	69.608858 sec	:   Mutex locked
DEBUG	69.608867 sec	: Setting radio/menu widget iso: 11 (800)
DEBUG	69.609042 sec	: Setting new configuration OK.
DEBUG	69.609050 sec	: Using internal bulb widget:bulb
DEBUG	69.609057 sec	: Setting toggle widget bulb: 1
DEBUG	69.627320 sec	: Setting new configuration OK.
DEBUG	69.627359 sec	: Exposure started
INFO	69.627370 sec	: Starting 1 sec exposure
DEBUG	69.627449 sec	: Time left: 1000
DEBUG	70.627648 sec	: Reading exposure...
DEBUG	70.627695 sec	: Time left: 0
DEBUG	70.627702 sec	: Closing shutter
DEBUG	70.627709 sec	: Using widget:bulb
DEBUG	70.627716 sec	: Setting toggle widget bulb: 0
DEBUG	70.644312 sec	: Setting new configuration OK.
DEBUG	70.644366 sec	: Exposure complete
DEBUG	70.648790 sec	: Unknown event.
DEBUG	70.651903 sec	: Unknown event.
DEBUG	70.655522 sec	: Unknown event.
DEBUG	71.714851 sec	: Event timed out #1, retrying...
DEBUG	72.773867 sec	: Event timed out #2, retrying...
DEBUG	73.829475 sec	: Event timed out #3, retrying...
DEBUG	74.887770 sec	: Event timed out #4, retrying...
DEBUG	75.944219 sec	: Event timed out #5, retrying...
DEBUG	77.001773 sec	: Event timed out #6, retrying...
DEBUG	78.059194 sec	: Event timed out #7, retrying...
DEBUG	79.116672 sec	: Event timed out #8, retrying...
DEBUG	80.175339 sec	: Event timed out #9, retrying...
DEBUG	81.232014 sec	: Event timed out #10, retrying...
ERROR	81.232088 sec	: Exposure failed to save image... Unspecified error

The entire INDI log (3.7MB) is attached
[indi_gphoto_ccd_23:38:04.txt](https://github.com/indilib/indi/files/1053488/indi_gphoto_ccd_23.38.04.txt)

knro, 2017-06-06:
Can you try to do the same thing with gphoto2 from the console?

justanenduser, 2017-06-06:
I am running gphoto2 2.5.11-1 and libgphoto2 2.5.14-1

I have run 10 configurations each one beginning with the camera powered off
and USB disconnected. Only 4 of them are successful.

Attached is my (hopefully understandable) summary.

-gary-

gphoto2 --bulb=2 --capture-image-and-download
capture=YES, download=NO, exposure time=UNKNOWN, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --capture-image-and-download
Bulb mode enabled (exposure time: 2s).                                         
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --capture-image --wait-event-and-download=2s
capture=YES, download=YES, requested exposure time=YES, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --capture-image --wait-event-and-download=2s
Bulb mode enabled (exposure time: 2s).                                         
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Event c201, Param1 ffffc001
Saving file as capt0000.arw                                                    
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Property d219 changed
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Event c202, Param1 ffffc001
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --capture-image --wait-event-and-download=5s
capture=YES, download=YES, requested exposure time=YES, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --capture-image --wait-event-and-download=5s
Bulb mode enabled (exposure time: 2s).                                         
Waiting for 5 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Event c201, Param1 ffffc001
Saving file as capt0000.arw                                                    
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Event c202, Param1 ffffc001
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property d21d changed
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --capture-image --capture-tethered=2s
capture=YES, download=YES, requested exposure time=YES, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --capture-image --capture-tethered=2s
Bulb mode enabled (exposure time: 2s).                                         
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Event c201, Param1 ffffc001
Saving file as capt0000.arw                                                    
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Event c202, Param1 ffffc001
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --capture-image --capture-tethered=2s
capture=YES, download=YES, requested exposure time=YES, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --capture-image --capture-tethered=5s
Bulb mode enabled (exposure time: 2s).                                         
Waiting for 5 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Event c201, Param1 ffffc001
Saving file as capt0000.arw                                                    
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Event c202, Param1 ffffc001
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property d21d changed
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --trigger-capture --capture-image-and-download
capture=YES, download=YES, requested exposure time=NO, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --trigger-capture --capture-image-and-download
Bulb mode enabled (exposure time: 2s).                                         
New file is in location /capt0000.arw on the camera                            
Saving file as capt0000.arw
Deleting file /capt0000.arw on the camera
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --trigger-capture --wait-event-and-download=2s
capture=YES, download=YES,  requested exposure time=NO, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --trigger-capture --wait-event-and-download=2s
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property 5004 changed
Saving file as capt0000.arw                                                    
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --trigger-capture --wait-event-and-download=5s
capture=YES, download=YES,  requested exposure time=NO, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --trigger-capture --wait-event-and-download=5s
Waiting for 5 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property 5004 changed
Saving file as capt0000.arw                                                    
UNKNOWN PTP Event c201, Param1 ffffc001
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Event c202, Param1 ffffc001
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property d21d changed
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --trigger-capture --capture-tethered=2s
capture=YES, download=YES, requested exposure time=NO, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --trigger-capture --capture-tethered=2s
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Event c201, Param1 ffffc001
Saving file as capt0000.arw                                                    
[gary@bigboote ~]$ 

gphoto2 --bulb=2 --trigger-capture --capture-tethered=5s
capture=YES, download=YES, requested exposure time=NO, program exits to command line
[gary@bigboote ~]$ gphoto2 --bulb=2 --trigger-capture --capture-tethered=5s
Waiting for 5 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Event c201, Param1 ffffc001
Saving file as capt0000.arw                                                    
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Property 5005 changed
UNKNOWN PTP Event c202, Param1 ffffc001
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property d21e changed
[gary@bigboote ~]$

justanenduser, 2017-06-10:
Some further exploration of gphoto2 on the command line...

--trigger-capture --capture-tethered=#s or --wait-event-and-download=#s
	this works but does NOT honor BULB (either camera exposure setting or the -B/--bulb option on command line
	it WILL work in other camera exposure setting (for my Sony A6000 that is 1/4000 sec ... 30 sec)
	use the SECONDS option in the command line and make it at least 7 seconds > camera exposure time

--capture-image-and-download
	this works but ONLY camera exposure setting OTHER than BULB (for my Sony A6000 that is 1/4000 sec ... 30 sec)

--capture-image -B/--bulb --capture-tethered=#s or --wait-event-and-download=#s
	this works
	use the SECONDS option in the command line and make it at least 7 seconds > camera exposure time

justanenduser, 2017-07-14:
The Sony A6000 uses LENR (long exposure noise reduction) by default. It has the effect of doubling the length of time it takes to save the image. If I turn LENR off indi_gphoto_ccd does manage to save and download the first image and the length of the exposure seems to have no effect. I can also cause the first image to be saved if I increase the timeout (gphoto_driver.cpp line 912) instead of turning LENR off, however as the exposure time increases I eventually hit the TIMEOUT failure.

knro, 2017-10-20:
How long would it take to save a 30-seconds exposure for example?

justanenduser, 2017-10-20:
I am now running gphoto2 version 2.5.15, libgphoto2 version 2.5.16.1 and
libindi version 1.5.0

If I turn LENR off libindi will successfully handle exposures from 1 second
to 60 seconds (I have tested these) and presumably any longer exposure.
If I turn LENR on (which doubles the amount of processing time done by the
camera) libindi fails (as reported by CCDciel) with any exposure longer
than 9 seconds.
Apparently the libgphoto2 driver for the Sony A6000 does not provide to
libindi any clearly identified status/signal to indicate when LENR is being
used. My "work-around" is to simply not use LENR.
If I use gphoto2 from the commandline I can specify a delay (to allow for
the increased time required by LENR) before downloading the processed
image.
I should probably have closed this issue.

closed 2017-10-20 by justanenduser