indilib/indi #260: CMake pthread lookup breaking build

TallFurryMan, 2017-06-18:
One commit breaks the build for me (Gentoo, gcc 4.9.4).
```
indi@astro1:~/indi_dev/indi$ git bisect bad
aee3c73c1278819c8a893cc27091b6eb229991c9 is the first bad commit
commit aee3c73c1278819c8a893cc27091b6eb229991c9
Author: Csaba Kertesz <csaba.kertesz@gmail.com>
Date:   Sat Jun 10 13:34:58 2017 +0300

    Add verbose warning and optimized debug/linker flags

:040000 040000 b9fccc43f30ab6b2eceb5931fb781c72e19b20f1 312da5aa41ff1597b76dbe189e0aae73581e4a92 M      libindi
```

PThread suddenly not found as a dependency.

```
+ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=RelWithDebInfo /home/indi/indi_dev/indi/libindi/
-- The C compiler identification is GNU 4.9.4
-- The CXX compiler identification is GNU 4.9.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - not found
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindThreads.cmake:223 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:201 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/indi/indi_dev/build/indi/CMakeFiles/CMakeOutput.log".
See also "/home/indi/indi_dev/build/indi/CMakeFiles/CMakeError.log".
```

I'll check how I can work around.
-Eric

TallFurryMan, 2017-06-18:
```
indi@astro1:~/indi_dev/indi$ cmake --version
cmake version 3.6.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

TallFurryMan, 2017-06-18:
Switch "-Werror" causes CMake 3.6.3's source snippet checking for pthread.h to fail.
The problem occurs because of the redefinition of _FORTIFY_SOURCE, defined internally by gcc 4.9.4 on Gentoo.

Often cited solution is to undefine "_FORTIFY_SOURCE" before defining it again on the command line. No very elegant, but effective.

I'll provide a pull request, merge as you wish.

TallFurryMan, 2017-06-18:
Offering to fix with PR #261.

kecsap, 2017-06-18:
Did you try to compile the bleeding-edge git? It should contain a fix already for your build error.

TallFurryMan, 2017-06-18:
I had a look. In my case cmake was not using optimization when checking for pthread, whatever the configuration. I'll double check but I think my pull request is still necessary for both libindi and 3rdparty. Of course, in my situation, _FORTIFY_SOURCE is predefined to 2 by gcc, so we loose nothing.

kecsap, 2017-06-18:
@TallFurryMan: Can you send me your cmake error log (CMakeError.txt or CMakeError.log) to my email. I am curious: csaba.kertesz@gmail.com

TallFurryMan, 2017-06-18:
Ah, just deleted the build tree to restart a dark library with bleeding edge.
The initial error was a #warning, stating _FORTIFY_SOURCE was redefined, as gcc 4.9.4 predefines it.
The second error was also a #warning, stating _FORTIFY_SOURCE had to be used with optimization, as cmake was not checking for pthread with that. I didn't test yet with the manual addition of -O1, but I think I remember it was not using the C flags. Probably pthread was a lone case.
I'll reproduce and let you know.

kecsap, 2017-06-18:
Now I see. Your PR really fixes the problem.

PR #261 (2017-06-18): Avoid using _FORTIFY_SOURCE on systems where compiler predefines it.
Testing for flag _FORTIFY_SOURCE with warnings as error to determine whether CMake will be able to test for library dependencies without optimization using the local compiler. Fixes #260 for CMake 3.6.3 with gcc 4.9.4 on Gentoo.
closed 2017-06-18 by knro