indilib/indi #263: make fails on indiccd.cpp using GCC 7.1.1 on Fedora 26

justanenduser, 2017-06-19:
There seems to a size disagreement between sender and receiver. 
The console fragment is as follows ... 

/home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp: In member function ‘bool INDI::CCD::uploadFile(CCDChip*, const void*, size_t, bool, bool)’:
/home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp:2436:6: error: ‘__builtin___snprintf_chk’ output may be truncated before the last format character [-Werror=format-truncation=]
 bool INDI::CCD::uploadFile(CCDChip *targetChip, const void *fitsData, size_t totalBytes, bool sendImage,
      ^~~~
In file included from /usr/include/stdio.h:939:0,
                 from /home/gary/Downloads/indi-master/libindi/libs/lilxml.h:66,
                 from /home/gary/Downloads/indi-master/libindi/indidevapi.h:97,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indibase.h:5,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/basedevice.h:21,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/defaultdevice.h:21,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.h:27,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp:25:
/usr/include/bits/stdio2.h:65:44: note: ‘__builtin___snprintf_chk’ output between 2 and 65 bytes into a destination of size 64
        __bos (__s), __fmt, __va_arg_pack ());
                                            ^
/home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp:2436:6: error: ‘.z’ directive output may be truncated writing 2 bytes into a region of size between 0 and 63 [-Werror=format-truncation=]
 bool INDI::CCD::uploadFile(CCDChip *targetChip, const void *fitsData, size_t totalBytes, bool sendImage,
      ^~~~
In file included from /usr/include/stdio.h:939:0,
                 from /home/gary/Downloads/indi-master/libindi/libs/lilxml.h:66,
                 from /home/gary/Downloads/indi-master/libindi/indidevapi.h:97,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indibase.h:5,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/basedevice.h:21,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/defaultdevice.h:21,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.h:27,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp:25:
/usr/include/bits/stdio2.h:65:44: note: ‘__builtin___snprintf_chk’ output between 4 and 67 bytes into a destination of size 64
        __bos (__s), __fmt, __va_arg_pack ());
                                            ^
/home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp:2436:6: error: ‘__builtin___snprintf_chk’ output may be truncated before the last format character [-Werror=format-truncation=]
 bool INDI::CCD::uploadFile(CCDChip *targetChip, const void *fitsData, size_t totalBytes, bool sendImage,
      ^~~~
In file included from /usr/include/stdio.h:939:0,
                 from /home/gary/Downloads/indi-master/libindi/libs/lilxml.h:66,
                 from /home/gary/Downloads/indi-master/libindi/indidevapi.h:97,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indibase.h:5,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/basedevice.h:21,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/defaultdevice.h:21,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.h:27,
                 from /home/gary/Downloads/indi-master/libindi/libs/indibase/indiccd.cpp:25:
/usr/include/bits/stdio2.h:65:44: note: ‘__builtin___snprintf_chk’ output between 2 and 65 bytes into a destination of size 64
        __bos (__s), __fmt, __va_arg_pack ());
                                            ^
At global scope:
cc1plus: error: unrecognized command line option ‘-Wno-unused-command-line-argument’ [-Werror]
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/indidriver.dir/build.make:279: CMakeFiles/indidriver.dir/libs/indibase/indiccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:663: CMakeFiles/indidriver.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

kecsap, 2017-06-19:
My PR will fix your problems: https://github.com/indilib/indi/pull/262

justanenduser, 2017-06-19:
The -Werror flag in line 93 of CmakeLists.txt is producing the compile failure I'm reporting here. 

    # Verbose warnings and turns all to errors
    SET(COMP_FLAGS "${COMP_FLAGS} -Wall -Wextra -Werror")

kecsap, 2017-06-20:
Please test the latest master. My PR was merged, all gcc-7 warnings are fixed or silenced. It should build fine for you.

justanenduser, 2017-06-20:
I am almost afraid to report this ...

Using a fresh copy from GitHub and the following command line issued from
the libindi subdirectory,
    cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local .

I get the following command line fragment,
    -- Performing Test COMPATIBLE_FORTIFY_SOURCE
    -- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
    -- Looking for pthread.h
    -- Looking for pthread.h - not found
    CMake Error at
/usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
      Could NOT find Threads (missing: Threads_FOUND)
    Call Stack (most recent call first):
      /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:377
(_FPHSA_FAILURE_MESSAGE)
      /usr/share/cmake/Modules/FindThreads.cmake:212
(FIND_PACKAGE_HANDLE_STANDARD_ARGS)
      CMakeLists.txt:220 (find_package)

From the CMakeError.log I see the following fragment,
    cc1: error: unrecognized command line option
‘-Wno-unused-command-line-argument’ [-Werror]
    cc1: all warnings being treated as errors

Looks like another troublesome warning.

HOWEVER...

Using a fresh copy from GitHub and the following command line issued from
the libindi subdirectory,
    cmake -DCMAKE_INSTALL_PREFIX=/usr/local .

Both cmake and make work as expected.

BUT ...

Trying to build the drivers from the 3party subdirectory using the
following command line,
    cmake -DCMAKE_INSTALL_PREFIX=/usr/local .

I get the following command line fragment,
    -- Performing Test ERROR_NAME_COMPILE
    -- Performing Test ERROR_NAME_COMPILE - Failed
    --  - 1.0.8 or older
    -- Looking for pthread.h
    -- Looking for pthread.h - not found
    CMake Error at
/usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
      Could NOT find Threads (missing: Threads_FOUND)
    Call Stack (most recent call first):
      /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:377
(_FPHSA_FAILURE_MESSAGE)
      /usr/share/cmake/Modules/FindThreads.cmake:212
(FIND_PACKAGE_HANDLE_STANDARD_ARGS)
      indi-sx/CMakeLists.txt:21 (find_package)

From the CMakeError.log file I see the following fragment,
    usr/include/features.h:373:4: error: #warning _FORTIFY_SOURCE requires
compiling with optimization (-O) [-Werror=cpp]
     #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
        ^~~~~~~
    cc1plus: all warnings being treated as errors

TallFurryMan, 2017-06-20:
I think each time you change CMakeLists, you should remove the build folder to be on the safe side. For the 3rdparty folder, the same change as in libindi was pushed and I suppose dust needs to settle on the ppa :)

kecsap, 2017-06-20:
Ok, the problem is two-fold here:
- The CMakeLists.txt in 3rd-party plugins did not contain the fix from @TallFurryMan for _FORTIFY_SOURCE.
- I think I was too strict to turn all compiler warnings to errors by default. It can create some hassle for people working on non-Debian systems and it is clear that we can't run tests on every system.

I created a new PR to solve all these problems:
1. Compiler warnings will not be turned to errors by default. Although I still would like to enforce it on Travis CI.
2. I added @TallFurryMan's fix to the 3rd-party's CMakeLists.txt.

PR: https://github.com/indilib/indi/pull/266

kecsap, 2017-06-20:
Ok, my PR merged. Please try the latest master again.

justanenduser, 2017-06-20:
Thank you. It works.
I'm deeply grateful for all of your assistance. This turned into a far
bigger hassle for all of you than I could have guessed.
Have a blessed day.  :-)

kecsap, 2017-06-20:
@justanenduser: You can bless my day if you close this issue. :-)

justanenduser, 2017-06-20:
You shall receive.  :-)

Thanks,
-Gary-

closed 2017-06-20 by justanenduser