indilib/indi #264: Strange offset in FITS images using V4L2 driver

TallFurryMan, 2017-06-19:
Although I don't know whether it is related to V4L2, I observe strange offset in the beginning of my FITS images captured with a DMK-21AU04-AS. I converted the following 5-second FITS darks into (reversed, sorry) PNG files, settings undefined pixels to white.

[image]
[image]

On the second dark, I noticed that the hot pixels were not located at the same place. There is a white offset at the bottom which is not on the first picture. Here's the top part of a longer exposure showing the problem as seen in the KStars viewer (transparent menu is visible on the left, sorry):

[image]

Given the fact that the CCD here is 640x480, it appears that the first pixel of the picture is at X=372,Y=1, that is, and offset of 372+640*1=1012 bytes. I see the same offset on all corrupted pictures.

The occurrence during my last session was low (1 dark out of 120 with various durations for instance), but curiously enough, this issue appeared far more often when using the focuser tab in Ekos. I see no obvious relation there, except maybe the fact that the focuser tab is using crop extensively. I tried to fiddle the V4L2 settings, but could not find a trigger anywhere. That could also be my hardware starting to die.

Could anyone try to reproduce this in order to gather more information?

knro, 2017-10-20:
I can't reproduce it here. Are you sure this was not some random transient issue? or is it persistent?

knro, 2017-12-04:
Is this still an issue? Please test and report back.

TallFurryMan, 2017-12-04:
We can close this I think. It's still a problem for me, no other report from other users, and I'm still "busy" working the MGen guider out. The cam works under Win10 with the manufacturer software. I'll keep that open in my own repo.

closed 2017-12-04 by knro