indilib/indi #265: Meridian flip remaining on same side of meridian

TallFurryMan, 2017-06-20:
My last session was pointing at NGC6888, and was managed by the Ekos scheduler. The plan had ~20 480-second exposure. Ekos did request a meridian flip around 2am, but somehow the mount simply adjusted its position without flipping. This morning, the mount was about to hit the tripod, and Ekos was still locked up waiting for the flip. Extract from the mount log:

```
2017-06-20T02:14:11: Telescope slew is complete. Tracking TRACK_SIDEREAL... 
2017-06-20T02:14:11: Iterative Goto (1): RA diff = 0.83 arcsecs DE diff = 0.01 arcsecs 
2017-06-20T02:14:11: Slewing to RA: 20:13:00 - DEC: 38:25:56 
2017-06-20T02:14:10: Slewing mount: RA increment = 24, DE increment = -1 
2017-06-20T02:14:10: Setting Eqmod Goto RA=20.6867 DE=37.0964 (target RA=20.2166 DE=38.4322) 
2017-06-20T02:14:10: Aligned Eqmod Goto RA=20.6867 DE=37.0964 (target RA=20.2166 DE=38.4322) 
2017-06-20T02:14:10: GOTO ALign Nearest: delta RA = 0.470111, delta DEC  = -1.335801 
2017-06-20T02:14:10: Starting Goto RA=20.2166 DE=38.4322 (current RA=20.6867 DE=37.0964) 
2017-06-20T02:12:41: Telescope slew is complete. Tracking TRACK_SIDEREAL... 
2017-06-20T02:12:41: Iterative Goto (1): RA diff = 0.21 arcsecs DE diff = 0.16 arcsecs 
2017-06-20T02:12:40: Slewing to RA: 20:13:00 - DEC: 38:25:58 
2017-06-20T02:12:40: Slewing mount: RA increment = 36, DE increment = 0 
2017-06-20T02:12:40: Setting Eqmod Goto RA=20.6868 DE=37.097 (target RA=20.2167 DE=38.4328) 
2017-06-20T02:12:40: Aligned Eqmod Goto RA=20.6868 DE=37.097 (target RA=20.2167 DE=38.4328) 
2017-06-20T02:12:40: GOTO ALign Nearest: delta RA = 0.470111, delta DEC  = -1.335801 
2017-06-20T02:12:40: Starting Goto RA=20.2167 DE=38.4328 (current RA=20.6868 DE=37.097) 
2017-06-19T22:58:30: Align: current point is 4 
2017-06-19T22:58:29: Align Sync: point added: lst=16.97932676 celestial RA 20.22050000 DEC 38.46850000 Telescope RA 20.69061134 DEC 37.13269947 
2017-06-19T22:58:29: Align Triangulate: number of faces is 5 
2017-06-19T22:58:29: Align Pointset: added point 4 alt = 53.9745 az = 87.2263 
2017-06-19T22:57:59: Telescope slew is complete. Tracking TRACK_SIDEREAL... 
2017-06-19T22:57:59: Iterative Goto (2): RA diff = 1.10 arcsecs DE diff = 0.01 arcsecs 
2017-06-19T22:57:58: Iterative goto (1): slew mount to RA increment = 1154, DE increment = 0 
2017-06-19T22:57:58: Iterative Goto (1): RA diff = 11.01 arcsecs DE diff = 0.01 arcsecs 
2017-06-19T22:57:46: Slewing to RA: 20:13:08 - DEC: 38:28:09 
2017-06-19T22:57:46: Slewing mount: RA increment = 65638, DE increment = -58621 
2017-06-19T22:57:46: Setting Eqmod Goto RA=20.6903 DE=37.1327 (target RA=20.2189 DE=38.4693) 
2017-06-19T22:57:46: Aligned Eqmod Goto RA=20.6903 DE=37.1327 (target RA=20.2189 DE=38.4693) 
2017-06-19T22:57:46: GOTO ALign Nearest: delta RA = 0.471373, delta DEC  = -1.336597 
2017-06-19T22:57:46: Starting Goto RA=20.2189 DE=38.4693 (current RA=20.8648 DE=39.4713) 
```

The log shows the last alignment on NGC6888 around 11pm, and the meridian flip around 2am. However I see no mention of the actual flip, if any is to be visible in that log. Effectively the flip was not done. I could see Ekos waiting for the flip (and a big problem with guiding at that specific moment, which I also need to debug):

[image]

Unfortunately KStars had no debug log set (configuration error from me), so I don't know what the engine was doing or expecting. So I'll have to reproduce the issue with a specific scenario.

Any idea what could lead to this situation?

knro, 2017-06-20:
The flip HA value was set to what? 0?

TallFurryMan, 2017-06-20:
Yes. NGC6888 was supposed to transit at 2:13am UTC, and the astro twilight was at 2:17am UTC. So I figured that would be a good opportunity to test the flip at the time of transit, without risking a hit into the tripod (though I was a bit optimistic: I expected more than 10cm before hit!). Note that to be on the "safe" side for the test, I also disabled the twilight check in the scheduler.

TallFurryMan, 2017-06-20:
This is a master tip build for indilib, and 2.7.8 KStars win32 from the release page (so I should probably update KStars).

knro, 2017-06-20:
The meridian flip should have timed out and exposure aborted, (after retrying a couple more times) but without Ekos logs not much can be said.

TallFurryMan, 2017-06-21:
Trying to reproduce. I thought I had succeeded, but instead got another issue :)
When the Ekos Scheduler finishes a job and shuts down INDI, if I reconnect INDI manually (clicking on the Start button) and restart a new Scheduler job, the job starts and says capture is in progress but nothing actually starts on INDI side (not a single INDI command sent).  Of course, meridian flip doesn't occur but nothing occurs at all in fact, mount keeps tracking and goes reverse. That might be related, but is definitely not the same scenario. So let's keep this one for another session.
Still trying to reproduce :)

TallFurryMan, 2017-06-21:
Small related issue reproduced, completely minor but fun when it happens:

```
017-06-21T08:04:58.328 - DEBG - V4L2 CCD :  "Capture of one frame (0 stacked frames) took 120.023275 seconds. "
2017-06-21T08:04:58.468 - WARN - Audio notification requested, but sound file from notifyrc file was not found, aborting audio notification
2017-06-21T08:04:58.492 - DEBG - fits MIN:  185  - fits MAX:  255  - pixel range:  70  - bin width  0.126582  bin count  554
2017-06-21T08:04:58.501 - DEBG - FITHistogram: JMIndex  0.0314366
2017-06-21T08:04:58.505 - DEBG - Capture:  "Received image 10 out of 10."
2017-06-21T08:04:58.508 - DEBG - Capture:  "Current hour angle 0.02 hours exceeds meridian flip limit of 0 hours. Auto meridian flip is initiated."
2017-06-21T08:04:58.509 - DEBG - ISD:Telescope: Sending coords RA  0.155126  DEC  29.1834
2017-06-21T08:04:58.530 - DEBG - EQMod Mount :  "Starting Goto RA=0.155126 DE=29.1834 (current RA=0.155126 DE=29.1834) "
2017-06-21T08:04:58.563 - DEBG - EQMod Mount :  "Aligned Eqmod Goto RA=0.155126 DE=29.1834 (target RA=0.155126 DE=29.1834) "
2017-06-21T08:04:58.564 - DEBG - EQMod Mount :  "Setting Eqmod Goto RA=0.155126 DE=29.1834 (target RA=0.155126 DE=29.1834) "
2017-06-21T08:04:58.605 - DEBG - EQMod Mount :  "Slewing mount: RA increment = -4511994, DE increment = 3048939 "
2017-06-21T08:04:58.809 - DEBG - EQMod Mount :  "Slewing to RA:  0:09:18 - DEC: 29:11:00 "
2017-06-21T08:04:58.869 - DEBG - Scheduler: Find next job...
2017-06-21T08:04:59.869 - DEBG - Scheduler: Checking Park Wait State...
2017-06-21T08:04:59.869 - DEBG - Scheduler:  "1 job completed."
2017-06-21T08:04:59.870 - DEBG - Scheduler:  "Scheduler complete. Starting shutdown procedure..."
2017-06-21T08:04:59.871 - DEBG - Scheduler: Checking shutown state...
2017-06-21T08:04:59.872 - DEBG - Scheduler: Starting shutdown process...
2017-06-21T08:05:00.868 - DEBG - Scheduler:  "Shutdown complete."
2017-06-21T08:05:00.870 - DEBG - Scheduler: Stopping Ekos...
```

Ekos runs a meridian flip before checking whether it was processing the last job of the scheduler. So I observed the flip, and Ekos shut INDI down while the flip was still in progress! Probably this would have been different if I had checked Park Mount for the end of the plan.
But this is clearly not INDI, so I'll post a bug at KStars instead for this very one.
Still trying to reproduce.

knro, 2017-06-21:
Yeah you're starting to get into the fringe cases that slipped past me. I'll take a look at them

knro, 2017-06-21:
I fixed the small issue regarding performing meridian flip when the job is already over. Will look into the other issue next.

TallFurryMan, 2017-06-22:
Awesome, I'll check this :)
Reproduced again something different, I'll describe it on kstars. Scenario is still short captures around meridian. This time, Ekos requested a slew between two captures, and the mount chose to run a meridian flip by itself. However Ekos did not wait for the end of the slew to restart the exposure. Then at the next exposure, Ekos did request a meridian flip, which did not occur, and waited for its end before restarting the exposure.

TallFurryMan, 2017-06-22:
OK, opened 381544 and 381545 at KDE Bugzilla.

closed 2017-09-17 by knro