indilib/indi #276: GotoNova: isSlewComplete times out

kbahey, 2017-06-29:
After the second slew, there are timeout errors, and the mount will only move a little bit then stop. I traced the problem down, and here is a patch to fix it. It is verified to work.

```
diff --git a/libindi/drivers/telescope/lx200gotonova.cpp b/libindi/drivers/telescope/lx200gotonova.cpp
index dc0aefa..657309e 100644
--- a/libindi/drivers/telescope/lx200gotonova.cpp
+++ b/libindi/drivers/telescope/lx200gotonova.cpp
@@ -201,11 +201,11 @@ bool LX200GotoNova::isSlewComplete()
     int nbytes_read    = 0;
     int nbytes_written = 0;

-    strncpy(cmd, ":SE?#", 16);
+    strncpy(cmd, ":SE?#", 6);

     DEBUGF(INDI::Logger::DBG_DEBUG, "CMD (%s)", cmd);

-    if ((errcode = tty_write(PortFD, cmd, 4, &nbytes_written)) != TTY_OK)
+    if ((errcode = tty_write(PortFD, cmd, 6, &nbytes_written)) != TTY_OK)
     {   

```

Please make the mount description as 'GotoNova Kit' or 'GotoNova 8400'

The reason is: the GotoNova name is used for various hand controllers, as well as the upgrade kit. The 8401 is the hand controller, and it can in theory be used for other Alt Az mounts, and 8400 is the upgrade kit's part number (the very first thing that iOptron made). 

So renaming it as I suggested above will avoid cases where people would use this driver for non-equatorial mounts and complaining.

kbahey, 2017-07-03:
Jasem pushed a patch that fixes the issue. 

No action needed code wise. 

However, the rest of the the naming would be nice to have.

knro, 2017-07-03:
The label was also updated.

closed 2017-07-03 by knro