indilib/indi #280: pyIndi-client build bug

gnthibault, 2017-07-03:
Hello,

First of all, I wanted to tahnk you for this wonderfull project. I am so enthusiastic about it, that I would like to dive a bit deeper in the project in the month to come.

I don't know if it is the best place to post this, but it looks like there is a compile error on my setup (anaconda) for the pyIndi-client:

indiclientpython_wrap.cpp:20962:50: error: ‘PyString_AsStringAndSize’ was not declared in this scope
         if (PyString_AsStringAndSize(obj1, &buf, &len) == -1)

Thank you in advance for your help, and let me know if there is a better place to report this problem.
Wish this awesome project a very long life !!!

geehalel, 2017-07-04:
Hi,
Did you install the python (2 or 3) development packages from your distribution ?
Anyway I will check this evening if there are issues with the latest libindi from git.

knro, 2017-10-07:
Closing due to no feedback.

closed 2017-10-07 by knro