indilib/indi #292: Auto gain setting for QSI camera is not working correctly

phomer60, 2017-07-09:
I have been running some tests of the gain setting for my QSI583, running under Ubuntu Mate on a Raspberry Pi 3.
I have found it is not setting the auto gain differently when binning higher than 1x1.
By reviewing the FITS Electrons per ADU I get the following results when looking at the preview images.
Bin High Low Auto
1x1   0.5    1.1   0.5 (This is correct)
2x2  0.5    1.1   0.5 (Should be 1.1)
3x3  0.5    1.1   0.5 (Should be 1.1)
4x4  0.5    1.1   0.5 (Should be 1.1)
It would also be useful to have a GAIN value as Nebulosity uses.

knro, 2017-07-09:
I updated the driver. Actually, the default was HIGH but in the driver it showed AUTO. So it will report the correct one in the next update.

knro, 2017-07-11:
Please test and report back.

phomer60, 2017-07-15:
This is partially fixed.
When the gain is set to Auto the images appear to have the high gain applied for 1x1 and the low gain applied for 2x2, 3x3 and 4x4. However the FITS header is not showing this correctly - it shows EPERADU as 0.5 in all cases whereas 1.1 should be shown for 2x2, 3x3 and 4x4.

knro, 2017-07-18:
That's what QSI SDK returns.. I contacted QSI to check their say on this.

knro, 2017-07-27:
well QSI did not respond back. Try communicating with them regarding this issue.

phomer60, 2017-07-29:
Jasem,

I have raised a support call with QSI showing them where the issue is. I assume we are unable to fix this as it is copyright software.

knro, 2017-08-29:
Any response from QSI?

phomer60, 2017-08-29:
James,

They were going to get their programmer to look into the problem but I have not hear back since.

I’ll send them a follow up email if I don’t hear back in this week.

Regards

Paul

phomer60, 2017-09-06:
QSI got back to me after I sent them a reminder, however, they did not read or understand the issue correctly so I have now sent them a more detailed explanation of the problem and I am waiting for a response.

phomer60, 2017-09-17:
After examining my more detailed description of the issue they have now responded as below. Not exactly what I was hoping for.
"The best we can suggest is to work around this issue in the application layer. We do not expect another release of the QSI APi software in to foreseeable future."

knro, 2017-09-17:
I suppose we could then just hardcode EPERADU value. Shall I use 1.1 if autogain is set and binning is larger than 1x1?

phomer60, 2017-09-17:
That is essentially what QSI should be doing, but if we can do it, then it is solved for me.

commit 19ce703c (2017-09-17): Setting electronsPerADU manually. This fixes #292
closed 2017-09-17 by knro