indilib/indi #293: Selection of Mount Control causes Kstars/Ekos to abort

phomer60, 2017-07-09:
Running Ubuntu Mate on Raspberry Pi 3 and mount is Losmandy Gemini.
When restarting Ekos the INDI servers are still running.
[image]
[image]
[image]

knro, 2017-07-10:
Please try PPA build from tomorrow and report if it works.

phomer60, 2017-07-12:
Tried again today and no difference I can see, it is still aborting completely out of kstars / ekos.

phomer60, 2017-07-13:
I have even tried this using the simulators and it fails, aborting back to the Unix shell.

knro, 2017-07-18:
That's usually an indication of missing QML files. Try running kstars from the console and check the error when it exists. It will probably complain about a missing QML module not being installed properly.

phomer60, 2017-07-18:
I always run kstars from a console via VNC as shown above. I tried running it with a keyboard and monitor and it works. The only error I see is libEGL warning: DRI2: failed to authenticate.

When running through VNC it is now showing as per the attached image.
[image]

knro, 2017-08-31:
Closing since not INDI bug.

closed 2017-08-31 by knro