indilib/indi #365: indi_gphoto_ccd and Nikon D810A

koichifun, 2017-09-05:
With the latest indiserver and Ekos (kstars 2.8.2), any setting properties on the INDI Control panel cannot be accepted.
I've checked this on RPi3 with Raspbian Jessie and the latest gphoto2, as well as with Ubuntu Mate running INDI server supplied from the repo, and kstars on Mac OS as a client.

knro, 2017-09-05:
screenshots? logs?

koichifun, 2017-09-05:
Hello, Jasem. The situation is the same as when you remotely examined my raspi in the last April.
When I try to set some parameters in the Control panel, pushing a button turns the indicator into yellow. The config seems to be written in the xml file (attached), while the value is not reflected in the Control panel. gphoto2 CLI works well showing summary and capturing an image. The gdb does not show anything, nothing is left in the logs of Ekos and indi (both attached).
After the yellow indicator is shown, I cannot stop indiserver from the Ekos panel (SS attached).
The capture module does not show and allow to set the dimension of the frame, and after hitting Preview, a raw file is left on the camera while the fit viewer does not start up.
[Ekos_log_05-23-54.txt](https://github.com/indilib/indi/files/1279141/Ekos_log_05-23-54.txt)

[indi_nikon_ccd_log.txt](https://github.com/indilib/indi/files/1279147/indi_nikon_ccd_log.txt)

[CLI_indiserver.txt](https://github.com/indilib/indi/files/1279144/CLI_indiserver.txt)

[image]
[image]
[image]

knro, 2017-10-20:
Maybe this was a temporary build issue? Do you still get it with the latest release?

koichifun, 2017-10-20:
Yes, I still have with the latest release, which I've just built from the repo. The kstars is also the latest version, 2.8.6, 2017-10-09. The libgphoto2 is also the latest 2.5.15.

knro, 2017-10-20:
You have the driver log?

koichifun, 2017-10-21:
Attached are the logs of indi_gphoto and ekos with D810 and D810A, respectively.
As before, once I pushed a button on the INDI control panel, the indicator turned into yellow, then any action is not be accepted. Then the INDI server cannot be turned off from Ekos, so I had to use the Web Manager or a kill command on the RPi.
The construction of D810A is very similar to that of D810, while several items differ, for example, D810A has M* mode, which allows longer exposures than 30sec.
Ekos seems to fail to receive/display the values of the setting, but I have no idea which causes it.

[indi_nikon_ccd_09:52:11._D810log.log](https://github.com/indilib/indi/files/1403997/indi_nikon_ccd_09.52.11._D810log.log)
[indi_nikon_ccd_10:07:43_D810A.log](https://github.com/indilib/indi/files/1403998/indi_nikon_ccd_10.07.43_D810A.log)
[ekos_log_18-52-06_D810.txt](https://github.com/indilib/indi/files/1403999/ekos_log_18-52-06_D810.txt)
[ekos_log_19-07-41_D810A.txt](https://github.com/indilib/indi/files/1404000/ekos_log_19-07-41_D810A.txt)

knro, 2017-10-23:
Could this "yellow" state be while the image is being downloaded to Ekos? or does it persist after the image is downloaded? The logs appear OK and do not show any crash.

koichifun, 2017-10-23:
The indicator continued in yellow, once I pushed any setting button on the control panel, even when I didn't capture anything. Hence, it has nothing to do with a download of a file. The situation seems not to be altered, since you logged-in my RPi and saw what occurs in the last April.

knro, 2018-05-04:
Can this be repeated still with latest INDI?

koichifun, 2018-05-04:
Yes. I've built libindi together with 3rdparty items about 2 hours ago. I use the latest kstars.
The indicators on the indi control panel turn in yellow and never accept any change, once I apply any setting. Now I cannot stop indi by pushing Stop INDI on the Ekos panel, after I see the yellow indicator.
As before, the setting xml file was updated by SAVE on the Option panel, while the indicator is in yellow. I attach that xml file and Ekos log.
BTW, in the older versions, the xml files were generated for camera models, but now the only Nikon DSLR_config.xml is made. I need to input the image setting at each time I connect a camera whose resolution differs from the previous one. Will you revert the manner to make the setting xml?

[Nikon DSLR_config.xml.txt](https://github.com/indilib/indi/files/1974195/Nikon.DSLR_config.xml.txt)
[image]
[image]
[log_20-00-21.txt](https://github.com/indilib/indi/files/1974196/log_20-00-21.txt)

knro, 2019-09-16:
Can you please test this again with latest drivers and libgphoto2?

koichifun, 2019-09-17:
Hi James,

I’ve been away from INDI for about half a year, so newly installed Ubuntu 18.04 on my RPi3B+ and the latest
Gphoto2 (libgphoto2 2.5.23) and built INDI from the source.

Before starting INDI, I tried the command gphoto2 in the terminal of Mac OS 10.14 (Mojave).
For example, "gphoto2 —auto-detect" returns

Model                          Port                                            
----------------------------------------------------------
Nikon DSC D810A                usb:001,006     

As for INDI, the result was the same as before.
That is, INDI recognizes D810A, but I cannot change/apply the setting on the control panel.
The indicators turned into yellow, but nothing has been applied.
No error log at all.

Attached is a screenshot of the control panel showing the yellow indicators.
This state continues more than 30 min., until I disconnect INDI server.

Regards,
Koichi

knro, 2019-09-18:
Thanks for the report. Please open this issue in the 3rdparty repo issues --> https://github.com/indilib/indi-3rdparty/issues

closed 2019-09-18 by knro