indilib/indi #381: Auto calibration for flats may get into an infinite loop

phomer60, 2017-10-01:
Raspberry Pi 3 running Ubuntu Mate with a recent version of INDI sometimes get into an infinite loop searching for a valid flats exposure time for my QSI camera. I use the recommended starting time as one second and set the target ADU as 40,000.
I have attached a copy of the log file showing it working for RGB but failing for L. It has also failed for the colour filters previously but after an update only the L now failed.

[flats_log.txt](https://github.com/indilib/indi/files/1346890/flats_log.txt)

knro, 2017-10-07:
Thanks for the report, but this is not an INDI issue. Please report it against KDE bugs.

closed 2017-10-07 by knro

phomer60, 2017-10-07:
Jasem,

So I can report this properly which KDE software does the auto calibration use?

Paul

knro, 2017-10-07:
It's KStars! so either post in the INDI Forums --> Ekos, or at bugs.kde.org