indilib/indi #393: indi_asi_ccd when using binnin- Preview - Live Video

Elfe, 2017-10-17:
Hi,

the indi_asi_ccd driver crashes with my ASI1600MM-C.
I connect up the camera in INDI, go to the CCD tab in Ekos.
Select Binning 4x4.
Do a Preview. Image comes up.
Select Live Video. Driver crashes.

[log_20-13-00.txt](https://github.com/indilib/indi/files/1392112/log_20-13-00.txt)

knro, 2017-10-18:
I can't reproduce that... can you get GDB backtrace?

Elfe, 2017-10-18:
Hi,

I did just look at the code and saw that the "static QHYCCD
*cameras[MAX_DEVICES];" and the for loops that go up to deviceCount might
cause problems.

looks like running
-snip-
int main ()
{
  int a[4];
  int i=0;

  for (i=0;i<10;i++) {
    a[i]=2;
  }

  return 0;
}
-snip-

knro, 2017-10-20:
Yes, this is merged thanks. But this is for QHY, how is this related to your ASI problem above?

pauledd, 2017-11-11:
I can reproduce that. ASI174Mini, take preview with 1x1bin ... ok, switch to 2x2bin and preview ...ok,
keep bin @ 2x2 and press live video ... indi_asi_ccd driver SIGSEGV.

gdb outp.:
```

#gdb -p 10606
...
Attaching to process 10606
Reading symbols from /usr/bin/indi_asi_ccd...done.
Reading symbols from /usr/lib64/libindidriver.so.1...done.
Reading symbols from /usr/lib64/libindiAlignmentDriver.so.1...done.
Reading symbols from /usr/lib64/libcfitsio.so.5...(no debugging symbols found)...done.
Reading symbols from /lib64/libusb-1.0.so.0...(no debugging symbols found)...done.
Reading symbols from /lib64/libz.so.1...(no debugging symbols found)...done.
Reading symbols from /lib64/libpthread.so.0...Reading symbols from /usr/lib64/debug//lib64/libpthread-2.25.so.debug...done.
done.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Reading symbols from /usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/libstdc++.so.6...(no debugging symbols found)...done.
Reading symbols from /lib64/libm.so.6...Reading symbols from /usr/lib64/debug//lib64/libm-2.25.so.debug...done.
done.
Reading symbols from /usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/libgcc_s.so.1...(no debugging symbols found)...done.
Reading symbols from /lib64/libc.so.6...Reading symbols from /usr/lib64/debug//lib64/libc-2.25.so.debug...done.
done.
Reading symbols from /usr/lib64/libnova-0.15.so.0...(no debugging symbols found)...done.
Reading symbols from /usr/lib64/libjpeg.so.62...(no debugging symbols found)...done.
Reading symbols from /lib64/libdl.so.2...Reading symbols from /usr/lib64/debug//lib64/libdl-2.25.so.debug...done.
done.
Reading symbols from /usr/lib64/libgsl.so.0...(no debugging symbols found)...done.
Reading symbols from /usr/lib64/libgslcblas.so.0...(no debugging symbols found)...done.
Reading symbols from /lib64/libbz2.so.1...(no debugging symbols found)...done.
Reading symbols from /lib64/libudev.so.1...(no debugging symbols found)...done.
Reading symbols from /lib64/ld-linux-x86-64.so.2...Reading symbols from /usr/lib64/debug//lib64/ld-2.25.so.debug...done.
done.
0x00007f2e2a31ee93 in select () at ../sysdeps/unix/syscall-template.S:84
84      ../sysdeps/unix/syscall-template.S: Datei oder Verzeichnis nicht gefunden.
(gdb) c
Continuing.
[New Thread 0x7f2e28db2700 (LWP 11961)]
[New Thread 0x7f2e27837700 (LWP 11962)]
[New Thread 0x7f2e262bc700 (LWP 11963)]
(gdb) [New Thread 0x7f2e2587c700 (LWP 12527)]
[New Thread 0x7f2e2507b700 (LWP 12554)]
[Thread 0x7f2e2507b700 (LWP 12554) exited]
[Thread 0x7f2e2587c700 (LWP 12527) exited]
[New Thread 0x7f2e2587c700 (LWP 13681)]
[New Thread 0x7f2e2507b700 (LWP 13691)]
[Thread 0x7f2e2507b700 (LWP 13691) exited]
[Thread 0x7f2e2587c700 (LWP 13681) exited]
[New Thread 0x7f2e2587c700 (LWP 14297)]
[New Thread 0x7f2e2507b700 (LWP 14306)]
[Thread 0x7f2e2507b700 (LWP 14306) exited]
[Thread 0x7f2e2587c700 (LWP 14297) exited]
[New Thread 0x7f2e2587c700 (LWP 14478)]
[Thread 0x7f2e2587c700 (LWP 14478) exited]
[New Thread 0x7f2e2587c700 (LWP 14654)]
[New Thread 0x7f2e2507b700 (LWP 14655)]

Thread 4 "indi_asi_ccd" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f2e262bc700 (LWP 11963)]
0x00007f2e2bc2805b in CCDChip::binFrame (this=0x55652a0e7060)
    at /home/paul/build/indi/libindi/libs/indibase/indiccd.cpp:322
322                                 accumulator += *(RawFrame + j + (i + k) * SubW + l);
(gdb) bt
#0  0x00007f2e2bc2805b in CCDChip::binFrame (this=0x55652a0e7060)
    at /home/paul/build/indi/libindi/libs/indibase/indiccd.cpp:322
#1  0x00007f2e2bcd06aa in StreamRecorder::uploadStream (this=0x55652a0f07d0)
    at /home/paul/build/indi/libindi/libs/webcam/v4l2_record/stream_recorder.cpp:271
#2  0x00007f2e2bcd030e in StreamRecorder::newFrame (this=0x55652a0f07d0)
    at /home/paul/build/indi/libindi/libs/webcam/v4l2_record/stream_recorder.cpp:214
#3  0x00005565291c0583 in ASICCD::streamVideo (this=0x55652a0e57e0)
    at /home/paul/build/indi/3rdparty/indi-asi/asi_ccd.cpp:1645
#4  0x00005565291c039d in ASICCD::streamVideoHelper (context=0x55652a0e57e0)
    at /home/paul/build/indi/3rdparty/indi-asi/asi_ccd.cpp:1610
#5  0x00007f2e2af00637 in start_thread (arg=0x7f2e262bc700) at pthread_create.c:456
#6  0x00007f2e2a327b9f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

```

maybe that helps somehow.

knro, 2017-12-03:
Please check against latest libindi 1.6.0 PPA build and report back.

pauledd, 2017-12-03:
just tested against git pull... for me it works, no more crashes, I can see 2x2 bin'ed video preview without any issues

closed 2017-12-04 by knro