indilib/indi #402: ST4 port is not recognized by second device connecting to INDI server

fsignoret, 2017-10-25:
Hello,

The issue started when PHD2 was changed last summer to check the camera capabilities before using the ST4 port.
I opened the following ticket to PHD2: https://github.com/OpenPHDGuiding/phd2/issues/646

But after some investigation, it appears that this is not linked to PHD itself: the interface displayed by the first client to connect is correct (6). The second client will then display (2) and not offer the ST4 capabilities.

My camera is a lodestar, I am using the latest versions of INDI/kstars on ubuntu+RPi.

On the PHD issue discussion, you will find screenshots and logs of the problem.
Please tell me if you need more information.

Thanks
Regards,
Florian

knro, 2017-10-25:
This issue is already fixed in INDI. Are you using INDI PPA packages?

fsignoret, 2017-10-25:
I guess we are talking about the version installed on the RPi?
It says "1.5.0". I installed it from here: http://indilib.org/download/category/6-raspberry-pi.html
How to use the PPA instead (on Raspbian)?

knro, 2017-10-25:
There is no PPA for Debian, only Ubuntu. You need to download again and install (it is updated daily).

fsignoret, 2017-10-25:
Thanks, it works now!
It was a bit unclear whether the package was updated, as it displays the same version.
Thank you very much!

closed 2017-10-25 by fsignoret