indilib/indi #412: indi_synscan_telescope serious sync problem

miserable-git, 2017-11-07:
After upgrading indi to 1.5.0-39 from 1.5.0-36 the synscan driver seems to have a mind of its own and nearly bent my telescope! 
If I try to sync from kstars or ekos it seem to set the synscan co-ordinates to a completely different point on the other side of the sky, and if it tries to track it can go way off limits
It worked fine in previous versions, I'm running fc26 i686.

kecsap, 2017-11-07:
I made the recent changes to the synscan driver and it works fine with my Skywatcher Virtuoso Alt/Az mount. Please provide the following information:
- Be sure that you updated the handset firmware to the latest (3.39 or 4.39 depending your Synscan version).
- What kind of mount do you use with the Synscan?
- What is your approximate geolocation?

miserable-git, 2017-11-07:
Hi,
I'm using synscan 4.39.04 on an EQ6-R mount. Perhaps it's an EQ specific issue?
Approx location 52  N  1 W

Cheers

kecsap, 2017-11-07:
Mostly likely it is a coordinate transformation issue.
What latitude is shown in the Mount Information tab on the Synscan tab -1 or something else? Is the longitude +52?

miserable-git, 2017-11-07:
Hi,
On the Site Management it shows my correct location, but not on Mount Information.
Definately some sort of coordinate translation going on.
For example, if I ask Kstars to sync it to Vega 18:37 38 48' it sends synscan to 9:56 66 27
Deneb at 20 42' 45 21' goes to 15 24 83 26'.
It doesn't seem to track the translated positions and they are moving over time.
Hope that means something to you!

kecsap, 2017-11-07:
I am on East Side thus it is most likely a coordinate transformation issue on the West side. I will properly test all possible combinations shortly, but I am imaging now. I think I will fix your problem already tonight. Why don't you use EQMod driver? I think it is much more feature rich than Synscan driver.

kecsap, 2017-11-08:
At least I found a problem in the location update which is fixed in this PR: https://github.com/indilib/indi/pull/413

I wonder is that what is the starting position when you start up your system? Is that different now what is used to be in 1.5.0-36?
For example, my Virtuoso Alt/Az mount has a start position Alt:Az -> 0:0. When I power up the system, the Synscan handset will send this position to KStars since it is an Alt/Az mount. How does your mount behave?

miserable-git, 2017-11-08:
Hi,
It starts up pointing at the pole. If I drive the mount from the hand controller Kstars follows it fine. It's only when Kstars sends a location to the controller that it all goes wrong. I'm a bit limited for time to do more testing today but maybe at the weekend I can do do a git bisect on it if you're still stuck.

MJP68, 2017-11-10:
My EQ8 did the same thing last night with the Eqmod driver installed. Tried to sync on Deneb, and the scope went and pointed way south below the horizon. My handset is the 3 series, 3.38 but I was using an eqmod cable, with stellarmate.

kecsap, 2017-11-13:
@MJP68 @miserable-git 
If the same problem happened with Synscan and EQMod drivers then something general can be wrong. I found some small issues during the weekend how the KStars shows the Synscan driver tracking status, but nothing serious. I could only get similar bad things what you experienced if the stored driver Configuration contained a wrong location. In that case, when the driver started, it passed this wrong location to the handset first, but KStars updates the location 1-2 second later to the good one and the new location was passed again to the handset. In this case, the handset went completely insane. Can you try the following?:

1. Start the Synscan driver with your mount.
2. Double-check if the location is good on the Site Managerment tab in the Synscan driver.
3. Save the configuration: Click Configuration/Save on the "Options" tab.
4. Disconnect and stop the INDI server. Power off and on again your telescope/synscan handset.
5. Start INDI server again and check if the handset works as it should.

Tip: Instead of plate-solving, just use the right click popup menu in KStars on the skymap to sync the telescope to a certain point. You can test without real sky.

There is one other thing what I don't understand right now. When the driver does a Sync with the handset, the mount should not move anywhere, first the telescope point is reset to the synced point and then the point is synced to the handset. But sometimes, I noticed that the mount moves after syncing without any reason although this behavior can be stopped by slewing the mount a bit before starting the first sync. I still don't understand which condition triggers this behavior.

miserable-git, 2017-11-14:
Hi, sorry for late reply, I tried that but no difference.

I did manage to do a git bisect on it though, hope this helps-

b5487f854a25ea93b53cdcb49d5cd3ab467bac3f is the first bad commit
commit b5487f854a25ea93b53cdcb49d5cd3ab467bac3f
Author: Csaba Kertesz <csaba.kertesz@gmail.com>
Date:   Mon Oct 23 01:47:43 2017 +0300

    Use Slew Goto instead of the normal goto (Synscan Mount)

:040000 040000 6e98b9de84db828f800a80a70a6737b2bb591dcc f404ced82f10615ef23c10a851ea27b9927a0ec3 M	libindi

kecsap, 2017-11-15:
This PR will replace the original Goto back for EQ mounts:
https://github.com/indilib/indi/pull/420

knro, 2017-12-04:
Is it this an issue? Please test and report back.

knro, 2017-12-10:
Please reopen if you have the issue again.

closed 2017-12-10 by knro