indilib/indi #424: Auto calibration on flats is causing kstars to abort

phomer60, 2017-11-19:
I am running Ubuntu Mate on a Raspberry Pi 3 with a QSI 583 main camera.
I have flats set to calibrate at 40,000 with 4,000 leeway.
It runs through the calibration without any obvious issues and then takes the first flat, downloading it successfully. It then aborts with a segmentation fault.
It is giving some messages about an audio file but I have no idea why.
If I turn off the calibration and manually set the time it works fine.
[image]

knro, 2017-11-19:
Not enough information there, can you run it via GDB?

phomer60, 2017-11-19:
Jasem,

Can GDB run kstars without compiling it with a -g option?

Paul

> On 19 Nov 2017, at 4:49:12 pm, Jasem Mutlaq <notifications@github.com> wrote:
> 
> GDB

knro, 2017-11-19:
If you compiled KStars yourself, you have to enable debugging via -DCMAKE_BUILD_TYPE=Debug. If you install from PPA, there are already debug symbols.

phomer60, 2017-11-19:
Jasem,

I am installing from PPA.

When I run it I am running it I get the following.

(gdb) run
Starting program: /usr/bin/kstars 

Program received signal SIGSEGV, Segmentation fault.
0x76fd9dde in ?? () from /lib/ld-linux-armhf.so.3
(gdb) 

Is it worth continuing?

Paul

knro, 2017-11-19:
Nope.. that's a problem with GDB on Ubuntu Mate RPI3. I have the exact same setup as you so I will run it today and see if I can reproduce the crash.

knro, 2017-11-19:
I just tested the exam same settings on StellarMate and I didn't get any crash. So we probably still need a way to figure out GDB.

knro, 2017-12-04:
Closing this since it is not an INDI related issue.

closed 2017-12-04 by knro

phomer60, 2017-12-07:
Since this appeared to be unrelated to INDI, I investigated further and found that I needed to install the following and reboot the Raspberry Pi.

sudo apt install phonon4qt5 phonon4qt5-backend-gstreamer phonon4qt5-backend-vlc
I assume some dependencies have been added.

Regards

Paul

phomer60, 2017-12-07:
Sorry, my last email was too soon.

I hadn’t realised that after the segmentation error the Flat Duration had reset to Manual and under that setting there is no issue.

So as long as you set a target ADU it happens.

Regards

Paul