indilib/indi #435: Proposal: Switch to "never develop in master" model

jochym, 2017-11-30:
I have a proposal @knro:
Recently we have multiple build breaks in master. This is very unfortunate for development - since developers cannot trust master to be a solid foundation to build on. And merging from upstream/master may break your branch in unrelated places. Large projects usually follow a "never develop in master" policy. This means that the progress in master is possible only by merge of the PR after successful CI build of  the PR. In this way the master is guaranteed to be at least aways building and when we extend our tests also passing the tests. This may slightly increase the load of maintainers (mainly @knro sadly) but I believe this will be made up with the time saved on fixing build errors in master. 
I am working in this way in other projects and after initial small growing pains it is very effective. Note that you can have long-living work-in-progress PRs which are automatically amended by new check-ins in the development tree and can use github's code review mechanisms for developers and maintainers cooperation. After a short while the safety net of always building master is very comforting. Furthermore it makes it easier for the release process. 
And except changing our practice and maybe a README we do not need to change anything.
What you say?

Blueshawk, 2018-02-28:
I'd like to second this. A development branch would be helpful and prevent lost time during the one night out of 20 we have clear sky. :dango:

Blueshawk, 2018-02-28:
I've started doing that too as some of my personal tweaks get undone by the daily ppa. (i.e. focus module limits, OnStep development). Lots of folks won't mess with compiling though and so, wanting to be up to date or get a bug fix will switch to Jasem's daily ppa which comes from main branch. 
Maybe it oculd be handled with a "stable" or "release" ppa option for new installs I.e. opera vs. opera-developer or something similar could be cooked up? 
Indi-full-developer vs. indi-full-release?

mikefulb, 2018-03-01:
I'd vote for slower and more methodical progress.

jochym, 2018-03-01:
@TallFurryMan  I would not propose anything which increases load of @knro . This must be as non-disrupting to our main developer as possible. What I propose is actually shifting the burden to contributors like me. The proposal is to have a golden rule: no PR may be accepted if it breaks CI.
For Jasem it would mean moving with his development to the separate branch and just merging to master when he is ready. This will hopefully  prevent our periodic CI breakages which require his (or somebody else's) attention.

TallFurryMan, 2018-03-03:
Do we know what was the exact cause of the issues? CI support temporarily broken so bypassed, or merges disregarding CI state? CI state is one thing, but software quality will be better if the code is tested by multiple contributors before being accepted. Which would mean for instance "indi" and "indi-bleeding" branches, non-final branch merges in "indi-bleeding" and clear test status on each of those running merges before approval for "indi", final merge and removal of the PR.

jochym, 2018-03-05:
We do not need to expand our system to the whole new level with multiple branches and multiple test procedures. Take the present case: https://github.com/indilib/indi/pull/536 . This PR has a breakage on clang4 pipeline due to the fli driver. But the PR has nothing to do with FLI driver! The PR author has probably no clue how to fix it and is left hanging without any plausible course of action - which is frustrating for contributors. Nobody likes to be powerless. The only resolution being the FLI maintainer (probably Jasem) will get around in his busy schedule and fix it. The present situation makes CI, maybe not useless, but much less valuable. The root cause is, probably,  that we have earlier accepted a FLI PR which broke the build. We would avoid this if we shift the responsibility of fixing the PR to the _author_ of the PR, since he/she will be in the position to fix it.
The simple rule I am proposing is: _nobody_ develops in master, nothing gets merged before passing CI. Note that it is quite easy to setup CI for your private fork - I can create a variant of CircleCI setup for any private fork and put it in the repo for the rest of us. The only difference for Jasem will be explicit merging of his own fork. This is additional workload but I believe it is quite acceptable comparing to the constant fixing of bad PR.

TallFurryMan, 2018-03-07:
I first read your point with Problem Report instead of Pull Request, which illustrates the two facets of those different change imports... I'm OK with this, although sometimes we should probably have Pull Requests part of a greater scheme, features discussed as forum topics for instance. This would allow better test coverage and focus from community, stretching the maturation period of the change before being merged.
+1 for never developing in master, as we could setup ppa build on master and CI on the integration branch.

jochym, 2018-03-07:
@TallFurryMan regarding branches etc. The feature branches are of course a great idea - in fact I am using it all the time and I have explicitly implemented support for single-driver branches in our CI (if the branch has name: drv_drivername only the drivername will be build in CI. Regarding the role of master - some projects work with the "always releasable master" - which is great but I do not think we are at that stage presently. But IMHO this is a worthy goal and I would strongly support going in this direction.

knro, 2019-09-18:
I believe we can revisit this proposal once we have more active developers that warrant the overhead associated with this development approach.

closed 2019-09-18 by knro