indilib/indi #437: Long queues on travis

jochym, 2017-11-30:
Recently travis is suffering really long queues and waiting times. I was thinking about splitting linux build  part to circleci which is much quicker. It will be a bit of an effort but we will have some redundancy in the CI department. What do you think @knro ?

knro, 2017-12-01:
Sure, let's give is a go.

jochym, 2017-12-01:
It turned out to be fairly easy. The config is different, but not that much, and I have reused travis scripts with just minimal changes. It basically works for default config. The variants (matrix) are not implemented yet but should be fairly easy. Check my fork's master  .circleci/ directory. 
There is also additional potential for optimization - the build runs in docker and we can build pre-configured image and avoid most of the setup time for the job.

BTW what is the reason for building everything three times (standard, debug, release)? This makes the build quite long and less useful for standard development cycle of committing often.

Shell I prepare a preliminary PR against the main repo?

jochym, 2017-12-02:
The PR (https://github.com/indilib/indi/pull/440) is almost ready - it builds on four platforms:

1. debian:stable

knro, 2017-12-03:
Great, I think maybe adding Fedora or Suse would be good to have. I am satisfied with the current structure now. Great work!

jochym, 2017-12-03:
Do you have a tutorial for compilation on fedora? I can make it myself but if you have one ready that would save me some work.

kecsap, 2017-12-04:
@jochym Yes, if we can build the sources on other Linux platforms with clang+gcc6+gcc7 configs, there is no need for these old trusty targets on Travis CI.

closed 2017-12-07 by jochym