indilib/indi #443: Install indi problem

Andreaweb80, 2017-12-03:
Hello, today i have a problem...
I've always installed all the indi lib and kstars with the following command:
sudo apt-add-repository ppa:mutlaqja/ppa
sudo apt-get install indi-full
sudo apt-get install indi-full kstars-bleeding

But today it response with an error:
I seguenti pacchetti hanno dipendenze non soddisfatte:
 indi-full : Dipende: libindi1 (>= 1.2) ma non sta per essere installato
             Dipende: libindi-dev (>= 1.2) ma non sta per essere installato
             Dipende: indi-bin (>= 1.2) ma non sta per essere installato
             Dipende: indi-eqmod ma non sta per essere installato
             Dipende: indi-sx ma non sta per essere installato
             Dipende: indi-sbig ma non sta per essere installato
             Dipende: indi-apogee ma non sta per essere installato
             Dipende: indi-gphoto ma non sta per essere installato
             Dipende: indi-qsi ma non sta per essere installato
             Dipende: indi-fishcamp ma non sta per essere installato
             Dipende: indi-maxdomeii ma non sta per essere installato
             Dipende: indi-asi ma non sta per essere installato
             Dipende: indi-aagcloudwatcher ma non sta per essere installato
             Dipende: indi-ffmv ma non sta per essere installato
             Dipende: indi-dsi ma non sta per essere installato
             Dipende: indi-qhy ma non sta per essere installato
             Dipende: indi-gpsd ma non sta per essere installato
             Dipende: indi-mi ma non sta per essere installato
             Dipende: indi-duino ma non sta per essere installato
             Dipende: indi-fli ma non sta per essere installato
             Dipende: indi-nexstarevo ma non sta per essere installato
             Dipende: indi-nexdome ma non sta per essere installato
             Dipende: indi-gpsnmea ma non sta per essere installato
E: Impossibile correggere i problemi, ci sono pacchetti danneggiati bloccati.

Seems a problem with the repo, but i don't know why...
Can someone help me?

MPaskin, 2017-12-03:
I have the same problem, only in English!

Andreaweb80, 2017-12-03:
So do you think it can be a repo problem?

MPaskin, 2017-12-03:
It's always worked before, so I think something has changed.

jochym, 2017-12-03:
Please, switch the report to english - this is the language the developers communicate.

knro, 2017-12-04:
Launchpad, which is used to build the PPA, was having technical problems yesterday. It appears today it is back again.

closed 2017-12-04 by knro