indilib/indi #444: EQMod - restore/load horizon data at startup

JohanBakk, 2017-12-03:
Feature request - could EQMod restore/load the horizon data at startup? Or alternatively be suited with a persistent preference to control the load on startup behaviour? 

Now the horizon data has to be loaded manually after startup. My scope is in a confined space and can not freely slew to all directions. Forgetting to load the horizon data may pose a danger to the mount.

Thanks in advance! 

Johan

geehalel, 2017-12-07:
Normally Horizon data are automatically loaded from `~/.indi/HorizonData.txt` if the file is here.
If you use another file name, this file name will not be restored between sessions (I think the Horizon part has been added before, or at the same time, the save/load feature in Indi). Is this your case ?

knro, 2019-09-18:
Thanks for the report. Please open this issue in the 3rdparty repo issues --> https://github.com/indilib/indi-3rdparty/issues

closed 2019-09-18 by knro

TallFurryMan, 2019-09-18:
@knro there's a current problem in horizon management which makes it unusable unfortunately. I'll open a new issue with information and possibly a PR.