indilib/indi #453: Could not find a package configuration file provided by "INDI" in Archlinux

JiangXL, 2017-12-09:
Hi! 
When I try to build and install QHYCCD driver to  INDI,  below error occurred. 
``` bash 
$ cmake -L
Could not find a package configuration file provided by "INDI" with any of
  the following names:
    INDIConfig.cmake
    indi-config.cmake
  Add the installation prefix of "INDI" to CMAKE_PREFIX_PATH or set
  "INDI_DIR" to a directory containing one of the above files.  If "INDI"
  provides a separate development package or SDK, be sure it has been
  installed.
```
INDI is install from binary package from Archlinu https://www.archlinux.org/packages/community/x86_64/libindi/files/.

Do it mean that INDI need to build from source to get  **INDIConfig.cmake** and 
**indi-config.cmake**.

Any idea is welcome!

knro, 2017-12-09:
You need to have cmake_modules directories somewhere. Try to clone the source and build libindi, libqhy, and then indi-qhy driver (in that order).

JiangXL, 2017-12-10:
It work!!!
Thanks, you are so nice :+1:

closed 2017-12-10 by JiangXL