indilib/indi #457: Eqmod and backlash after connection Indiserver version 1.6

azwing, 2017-12-12:
Hi,

When I use the eqmod driver I experience a strange behaviour after connection.
When I try to start tracking of other movement, the driver initiates a backlash compensation with a huge amount of microsteps leavig the mount to turn for a long time.

If I first set the backlash to 1 this does not happen.

Is this just a normal behaviour or a bug?

geehalel, 2017-12-19:
Hi,
I had the same issue after recompiling from scratch, it seems that uninitialized member variables are set to something not false by the compiler, don't know why. I fixed it but backlash does not work with the simulator, I believe it was working before. So don't use backlash with the simulator for the moment.

knro, 2018-10-26:
Is this still a problem in GIT? IIRC, all variables were initialized in EQMod.

closed 2018-12-17 by knro