indilib/indi #462: Locking filter when focussing is not working

phomer60, 2017-12-15:
When I lock the Luminance filter for focussing and I am imaging using a Ha filter and it does a focus run between images it it not changing the filter for the focus run. It ends up not having a star to focus on and attempts to focus on noise before sometimes thinking it is successful and other times giving up. Either way the rest of the night is lost.
[image]

knro, 2017-12-16:
This is not an INDI Bug, it's KStars/Ekos. Please file those bugs over at bugs.kde.org

Notwithstanding that, this is already fixed in KStars 2.8.9 which is due out later today.

closed 2017-12-16 by knro

phomer60, 2017-12-16:
Jasem,

Unless I am missing something on the INDI web site, it would probably be useful to have somewhere in the support area which describes which types of things are INDI, Ekos or Kstars as it may reduce the number of misdirected queries.

Regards

Paul

knro, 2017-12-16:
Sure. I'll try to conjure something. Basically, if the _driver_ is not working then it is INDI. For anything else it's the client fault (in this case KStars).

phomer60, 2017-12-21:
Jasem,

I am wondering if this bug fix has has actually been released, as it still seems to be occurring.

The kstars 2.8.9 build is showing as 2017-12-16T11:27:07Z

Regards

Paul

knro, 2017-12-21:
Please file a bug at bugs.kde.org