indilib/indi #465: Apogee 3rd party drive

deisenlord, 2017-12-19:
I built the lasted version of indi last night under OSX using the build_kstars.sh script.  I then tried to build the Apogee drivers.  There are a few simple problems that prevent libapogee from building under
OSX El Capitan 10.11

1).The use of the <tr1/memory> header file, which can be replace by simply <memory>
2).The use of an incompatible version number during linking.

I was able to work around these and the driver appears to work fine.   Looking at the Apogee site, now Andor, the latest libapogee version 3.0.3234 source contains these problem as well.  That said poking around inside their OSX 10.9 dmg archive the source does not contain these issues.   

It sounds like some conditional compilation is required to distinguish between linux and OSX.    Is this something I can help fix and who/where would be the correct place. 

Thanks, David
Minneapolis

knro, 2017-12-20:
Hi David,

Do you think you can submit a PR to make Apogee build on Mac?

deisenlord, 2017-12-20:
Good idea.  They seem very slow to respond these days but I will report back.
David

knro, 2017-12-20:
I meant submit a Pull-Request here against libapogee with your changes that made it work under Mac?

deisenlord, 2017-12-20:
Aha, I can do that as well.   

David

rlancaste, 2018-01-01:
I believe it is all done and apogee should work now, but of course I don't have one to test.  I will make a dmg in the next couple of days.  If you can test it beforehand, that would be very helpful

deisenlord, 2018-01-01:
I did a complete build a couple of days ago and it cleanly built a DMG for me for me with the apogee drivers.   I haven't tested it yet but will and report back, don't expect any problems.

Thanks!

David

rlancaste, 2018-01-01:
Hi David,

It is possible you did that build prior to my adding the environment variable and the firmware files into the KStars app.  I didn’t get a chance to do that until about 2 days ago.  A good test would be to rename the folder with the firmware files in it at /usr/local/etc/Apogee and see if the camera still works when you run it locally in KStars.

Thanks,

Rob

deisenlord, 2018-01-01:
Will check Rob

Sent from a black rectangle made by a company named after a fruit.

knro, 2018-01-14:
Another issue is that of documentation. Currently, there is barely any here: http://www.indilib.org/devices/ccds/apogee-ccd.html

Can you please submit screenshots and some documentation about the driver? We would really appreciate that so that prospective users can find information on the driver and any quirks they have to deal with on the official INDI devices documentation page.

You can see sample documentation here: http://www.indilib.org/devices/ccds/gphoto.html

deisenlord, 2018-01-14:
Sure thing Jasem, I'll take a look and put something together.

deisenlord, 2018-01-31:
Jasem, attached is a first cut at the Apogee documentation.  Pls let me know if there is more I can do in this regard.   David

[apogee-docs.is-a-zip.zip](https://github.com/indilib/indi/files/1682729/apogee-docs.is-a-zip.zip)

knro, 2018-02-08:
Thanks that is sufficient for now!

closed 2018-02-08 by knro