indilib/indi #489: Inconsistent Debug Properties behavior

Modiug, 2018-01-03:
System + Version: Ubuntu on RasPi3, updated today (2018-01-03), but also seen on Arch Linux with an older build.

To reproduce:
- Start Kstars
- Toggle Ekos
- Choose Profile "Simulators" and "Start INDI"
- In INDI Control Panel, goto CCD Simulator and click "Connect"
- Then goto Options tab and click Debug Disable
The Debug property switches back to Enable and the additional Debug Properties disappear and re-appear immidiatly (further down).
Hitting "Disable" again repeats the disappear and re-appear.

When you restart INDI and do a "Disable" on the Debug property (CCD Simulator) it stays disabled.
However, when you "Connect" the CCD Simulator, then go to Options and enable the Debug property again it stays enabled and the additional Debug Properties appear. Trying to disable now brings back the previous behavior (flashing to disabled but immediatly going to enabled again)

Regards
Guido

knro, 2018-01-03:
This is not an issue with INDI, it's with KStars. Also, it's not a bug, it's a feature! The reason is that CCD was selected in "Logs" button so it is _always_ enabled. Basically, with the logs button, you control the debugging behavior or INDI devices and not via INDI control panel directly. If you want it off, uncheck it in the Logs button window.

closed 2018-01-03 by knro

Modiug, 2018-01-03:
I just saw that I had Logging for CCD drivers enabled in Kstars->Ekos->Logs->Logging.
Therefore the default for CCD drivers was Debug enable.
The above described behavior is just inverted if it is not enabled.

Ah, OK, just saw your above message coming in while I was typing this. Quick reaction :-)
Hmmm, still there is this different behavior whether you have Connected the driver first or not.