indilib/indi #495: Hitecastrodc focuser driver in ekos/kstars

Robair996, 2018-01-10:
[log_17-12-31.txt](https://github.com/indilib/indi/files/1620598/log_17-12-31.txt)
Hi everyone,

I have a problem with my Hitecastro DC focus controller 2.
The idea is to use it with Kstars on a Mac but for some reason, although an INDI driver is installed for the Hitecastro it doesn’t seem to be able to connect.
Works fine manually. My motorfocusser is a JMI motofocus.
It also works fine under ASCOM with Backyard EOS and APT 3.5 (Astro Photography Tool) so I don't think there is a problem with the hardware. The same error when I run INDIserver and INDI control panel.
Please see attachments included (and the kstars/ekos logfile .
Thanks a lot for any help.
Kind regards.

[image]
[image]
[image]

knro, 2018-01-12:
Can you test this under RPI3 or Linux? Same issue?

Robair996, 2018-01-12:
Fear not, don’t have either.  Meanwhile I have been receiving additional info.

Just received a reply from a certain Kirkham Andy who apparently wrote the initial indi driver for the earlier version of the HitecAstro DC focuser controller. I quote him: 

" You need to ask the manufacturer for the USB "vendor id" and the "product id".

Now, copy /usr/lib/udev/rules.d/99-hitecastrodcfocuser.rules to /usr/lib/udev/rules.d/99-hitecastrodcfocuser.2rules

Edit /usr/lib/udev/rules.d/99-hitecastrodcfocuser.2rules and put in the vendor and product id.

Plug in the device and it should now become visible in your system. It's just really a virtual serial port.

--Andy "

 
I am not sure what to do with that. Is this a location on my Mac or a command line in the original HitecastroDC driver? I don't know.  I have been searching for this on the harddisk of my mac but without luck. I did find the location of the  indi driver though.

It probably means something to you.  

Meanwhile I have mailed HitecAstro company for the USB vendor ID and the product ID.

knro, 2018-01-12:
His reply is Linux-specific. udev rules are not applicable to Mac. Will have to research more regarding HID on Mac as I use Linux and don't know much about Mac-specific behavior.

Robair996, 2018-01-12:
OK, I see. I don’t know who wrote the actual HitecastroDC driver which is integrated in the indilib for OSX.

Thanks ahead,

Robert

Robair996, 2018-01-12:
Hi Jasem,

I managed to find the required hardware ID’s myself by connecting my HitecastroDC controller 2  to a windows machine.  
The VID number would be 04D8   and the PID I believe to be F53A.

I found and understood that I have to change some files in kstars  which I did/
Cmakelist.txt
drivers.xml
99-hitecastrodcfocus.rules
 hitecastrodcfocuser.cpp

I changed the PID to what it should be (according to the driver installed in Win 8.1) but unfortunately no change!  :-(

I saved all the changes and restarted Kstars as well as my focus controller.

What am I doing wrong?

Greetz,

Robert

Robair996, 2018-01-12:
Are there any files I have to do with CMake 2.8.12.2?  Is this a compiler?

Robert

knro, 2018-01-12:
As I said before, the udev rules are for Linux only, they have no effect on Mac. I added your PID to hitech focuser now, but you will need to compile INDI from source to be able to test that.

Robair996, 2018-01-12:
Thanks, 
How do I go about comiling. Which files and what program do I use for that? 
Haven’t done that before, toral newbie here.

Robair996, 2018-01-12:
Quite a steep learning curve for me at my 60 years of age and with no programming experience  I must say. :-)

I searched for what you refer to as the source code which I believe to have found at 

https://download.kde.org/stable/kstars/kstars-2.9.1.tar.xz.mirrorlist 

Meanwhile I also have a program called Cmake which I understand to be a compiler.

I had a go at “compiling” without success I’m afraid. I got errors of course. See log and screenshots.

I am trying. ;-)

Regards,

Robert

> 
> His reply is Linux-specific. udev rules are not applicable to Mac. Will have to research more regarding HID on Mac as I use Linux and don't know much about Mac-specific behavior.

# This is the CMakeCache file.
# For build in directory: /Applications
# It was generated by CMake: /Applications/CMake 2.8-12.app/Contents/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Build KStars Lite
BUILD_KSTARS_LITE:BOOL=OFF

//Path to a program.
CCACHE_FOUND:FILEPATH=CCACHE_FOUND-NOTFOUND

//Enable ccache support
CCACHE_SUPPORT:BOOL=OFF

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: None(CMAKE_CXX_FLAGS or
// CMAKE_C_FLAGS used) Debug Release RelWithDebInfo MinSizeRel.
CMAKE_BUILD_TYPE:STRING=

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler.
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

//Flags used by the compiler during all build types.
CMAKE_CXX_FLAGS:STRING=

//Flags used by the compiler during debug builds.
CMAKE_CXX_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during release minsize builds.
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the compiler during release builds (/MD /Ob1 /Oi
// /Ot /Oy /Gs will produce slightly less optimized but smaller
// files).
CMAKE_CXX_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
[... 264 lines cut ...]
//ADVANCED property for variable: CMAKE_SKIP_INSTALL_RPATH
CMAKE_SKIP_INSTALL_RPATH-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_SKIP_RPATH
CMAKE_SKIP_RPATH-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS
CMAKE_STATIC_LINKER_FLAGS-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_DEBUG
CMAKE_STATIC_LINKER_FLAGS_DEBUG-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_RELEASE
CMAKE_STATIC_LINKER_FLAGS_RELEASE-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STRIP
CMAKE_STRIP-ADVANCED:INTERNAL=1
//Suppress Warnings that are meant for the author of the CMakeLists.txt
// files.
CMAKE_SUPPRESS_DEVELOPER_WARNINGS:INTERNAL=FALSE
//uname command
CMAKE_UNAME:INTERNAL=/usr/bin/uname
//ADVANCED property for variable: CMAKE_USE_RELATIVE_PATHS
CMAKE_USE_RELATIVE_PATHS-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_VERBOSE_MAKEFILE
CMAKE_VERBOSE_MAKEFILE-ADVANCED:INTERNAL=1

knro, 2018-01-13:
Yeah I think we need a simple guide for MacOS users to just compile INDI. Will see if we can make such a guide. The current guide is for Linux.

Robair996, 2018-01-13:
I appreciate your understanding. ;-)

Greetz,
Robert

Robair996, 2018-01-28:
Hi Jasem,
Thanks for Kstars 2.9.2  I just tried it out and send you some feedback. The Hitecastro focuser controller indicates as being connected but there is no reaction when issuing commands (focus in or out from Ekos)
[log_20-50-16.txt](https://github.com/indilib/indi/files/1671650/log_20-50-16.txt)
[log_22-48-25.txt](https://github.com/indilib/indi/files/1671651/log_22-48-25.txt)

.
Also the Killall PTP feature doesn't seem to work ok. Although the PTP feature pops up to allow me to kill all apps connecting to the Canon DSLR program for some reason Ekos can not connect to the DSLR. The command in "Killall PTPCamera" in Terminal doesn't help either.

Robair996, 2018-01-28:
Sorry, inadvertently closed the topic.

knro, 2018-01-29:
I will try to contact HiTechAstro for more information on this. Thank you for your patience.

Robair996, 2018-01-29:
Hi Jasem,
On the contrary, thank you for all the effort put in a free program.

I have the the Hitecastro DC2 Focuser controller program which only exists in a Windows version but I am not sure whether you have means to access the code. I fear it is not free source.

There is one feature which is very helpful and that is a choice of focuser speed between 0 and 100%. It clearly speeds up the DC motor so that large focus position changes do not take so long.
I have not seen that in Kstars/Ekos. 

Kind regards,

Robert

Robair996, 2018-02-06:
Hi Jasem,

Where you able in obtaining further info on  the Hitecastro focuser controller?

Greetz,

knro, 2018-02-06:
I am waiting for a physical unit to be shipped so I can test directly, so I'll update this once I have any further news.

Robair996, 2018-02-06:
Great, Appreciated. 
Just to make sure, the Focuscontroller I ave is the Hitecastro DC focus controller 2.
They have different controllers at HitecASTRO.
I enclosed a picture.

Kind regards,

Robair996, 2018-02-26:
Hi Jasem,

Any luck with the Hitecastro DC Focus controller 2 driver?

Kind regards,

Robert

knro, 2019-01-27:
Great, please submit a PR! Looks like most users have controller 2

closed 2019-01-27 by knro