indilib/indi #500: QHY doesn't stop with >5 seconds exposures

SquareBoot, 2018-01-13:
Hello everyone,
Using the current `indilib` version I can't take photos that are 5 seconds or more long using a _QHY5LII-M_ on a _Xubuntu 16.04 32-bit_ computer. Everything is up-to-date. On the other hand, my _Ubuntu_ and _Kubuntu 17.10 64-bit_ computers work fine using the same version of the INDI drivers. It isn't a power supply problem.
When the problem occurs, the photo doesn't end and the indicator in the control panel remains yellow (using both KStars installed on Xubuntu and the one on Ubuntu via INDI Web Manager); then I must close the Web Manager and KStars to stop the photo. The zombie `indi_qhy_ccd` and `indiserver` processes will continue working until killed. `fxload` is working correctly, so the firmware is loaded.
No way to get some useful error logs from `dmesg`, the INDI control panel, the Web Manager (using `--verbose`) or from the remote KStars client: INDI simply sais that everything is working fine, but when I take a longer photo it halts. It isn't a bad PPA build: the same problem is occurring in various updates.
I also tried building `libindi` from the repo from commit `82499d580f2dd529d4571a9ea7a1ac857357c223` (a quite old `libindi 1.6.0` version that used the QHY SDK v. 1.0.1, supporting 32-bit PCs), from `cc83b1a703f1f6353f58467a2904ee40c4f37b4b` (when @knro added `qhycam.h`) and even from v. 1.5.0. The first attempt (compiled with all the 3rd party drivers) was unable to detect the camera although`fxload` loaded the firmware; the second one (compiled with only `indi_qhy_ccd`, `fxload` and `libqhy`) reported the same error. While compiling v. 1.5.0 `CMake` reported that I have some missing dependencies, which are, however, unavailable in Ubuntu 16.04, so I was unable to test it. Please note that the current QHY SDKs are only `x86_64` compatible: if you take a look at their repo, only `qhyccdsdk-1.0.8-Linux-Debian-`**x86_64**`.tar.gz` is available at the current commit. I didn't install any driver from the QHY official repo before compiling v. 1.6.0. The same problem occurred after installing them.
Another user reported the same problem on his first Raspberry Pi 3 using INDI 1.6.0, while his second Pi 3, with v. 1.5.0, was working fine.
I apologize for the long description, but it's a summary of the topic that can be found [here](http://indilib.org/forum/ccds-dslrs/2945-qhyccd-driver-can-not-expose-more-than-5-seconds.html?start=36) (unsolved).
Kind regards,
SB.

knro, 2018-02-08:
So this is only a 64bit vs 32bit issue? QHY SDK was just updated yesterday, can you try to see if you still have issues?

SquareBoot, 2018-02-08:
I haven't used my QHY for a while and I've just found that it doesn't work for >= 4 seconds photos on both PCs, before and after updating INDI.
Kind regards

knro, 2018-03-16:
It should work now. Please test.

SquareBoot, 2018-03-17:
Yes, it works, even with 60 seconds! I'll test the camera on the small computer as soon as it'll finish downloading updates. Feel free to ask me for further tests someday.
Thanks,
SquareBoot

closed 2018-03-17 by SquareBoot