indilib/indi #503: Lacerta MGen not provided by indi-full

TallFurryMan, 2018-01-19:
Following discussion threads at http://indilib.org/forum/ekos/2999-mgen-driver.html and http://indilib.org/forum/ccds-dslrs/1169-lacerta-mgen-support.html, it appears the Lacerta MGen driver is properly built as part of verification builds of the github repository, but is not provided as an executable by any PPA package.

Would it be possible to include this driver in what the PPA provides? What is the procedure?

knro, 2018-01-19:
It could be added to the PPA, but since Launchpad is still very limited, we need to wait.

TallFurryMan, 2018-01-20:
I'm puzzled by launchpad. There seems to be many "old" builds waiting in the queue. For instance, "indi-dsi" has 8 packages built for various distros and archs, but still, there are 46 builds waiting in line, each making the previous build immediately outdated...

knro, 2018-01-21:
Yeah I just asked Launchpad to batch-cancel them since they're over 1000+. I also added INDI MGEN debian files but waiting for this to sort out before I trigger the build. Or it will be triggered automatically tomorrow.

TallFurryMan, 2018-01-22:
I was able to retrieve `indi-mgen` for the Lubuntu 17 I was using. I probably selected the wrong solution, but I found I had to do a system upgrade from 17.04 to 17.10 before the driver would appear in the list in Synaptic. Indeed, there is no package provided for 17.04 on the PPA, only for 18.04, 17.10 and 16.04, which can appear strange somehow. Anyway, I think the ticket can be closed, thanks!

closed 2018-01-22 by TallFurryMan

TallFurryMan, 2018-01-23:
Just a side note about `indi-full` still at 1.4~201710281604 and not embedding `indi-mgen`, just in case there is plan to update that virtual package. Not reopening because fetching `indi-mgen` is an easy (though not documented) workaround.

Suggested addition to http://indilib.org/download/ubuntu.html:

> 4. To Install specific INDI drivers by package names:
> `sudo apt-get indi-eqmod indi-duino indi-mgen indi-nexdome`